Ultimaker/Cura #6191: [4.2.1] Slicer inserts M141 after start g-code

bastion-0078, 2019-08-11:
**Application version**
Cure 4.2.1

**Platform**
Windows 10 Pro x64

**Printer**
Anycubic Predator  (with Duet Maestro)

**Reproduction steps**
Slice a file with a RepRap Printer. 

**Actual results**
Slicer inserts a M141 S28 command after start g-code. 
**Expected results**
Command should not be inserted

**Additional information**
M141 is used in the Duet firmware to set a chamber temperature. Only an error message is displayed, but the print is not affected. But the problem doesn't always seem to occur.

rburema, 2019-08-13:
Just to confirm I understand: The problem is that the Anycubic Predator doesn't support buildvolume temperature (commands), and you'd therefore like the M141 command not to happen in that case.

ksihota, 2019-08-13:
I see the same behaviour right after my Start GCode on my MPMD. I just noticed it the other day.
v4.21 inserts:
M141 S28
G92 E0
G92 E0

4.1 inserts:
G92 E0
G92 E0

fieldOfView, 2019-08-13:
On top of the sidebar, in the Search field, type "Build Volume Temperature". Change the value of the setting that shows up to 0.

fieldOfView, 2019-08-13:
I really dislike that the somewhat arbitrary value of 0 is chosen to not insert a value. Not only could 0 be a valid temperature for some printer (eg printers that print perishables instead of plastics), but this is inconsistent with how heated beds are handled. I would expect there to be a `machine_heated_build_volume` setting to mirror the `machine_heated_bed` setting.

bastion-0078, 2019-08-14:
> Just to confirm I understand: The problem is that the Anycubic Predator doesn't support buildvolume temperature (commands), and you'd therefore like the M141 command not to happen in that case.

No, the Predator does not support this function. But I replaced the standard controller board with the Duet Maestro, which can be used in a variety of devices. CNC milling, laser-based cutting and engraving machines and also 3D printers. Therefore the M141 command is part of the RepRap command set. I'm just surprised that CURA inserts this command even though I didn't define a chamber temperature anywhere. If I set the temperature to zero I also get this error message, because zero degrees Celsius can be a plausible temperature.

fieldOfView, 2019-08-14:
Setting it to the “magic” value 0 should prevent the command from being inserted.

Ghostkeeper, 2019-08-14:
The somewhat arbitrary value of 0 was chosen just because it is consistent with the way that the heated bed temperature is used. Setting a value of 0 for the bed temperature will cause it to not insert the bed temperature command. But I agree that having a `machine_heated_build_volume` machine setting makes sense.

GregValiant, 2023-09-15:
Is this still a problem or does the "Heated Buiild Volume" setting in the Machine Settings cover it?  If it does then can this be closed now?

closed 2023-09-29 by github-actions[bot]