Ultimaker/Cura #6195: [4.2.1] Color problem with Settings-GUI on linux and dark KDE/QT-theme (Breeze dark)

esopalumpa, 2019-08-12:
**Application version**
4.2.1 probably older versions too.

**Platform**
Linux, KDE, Theme “Breeze Dark”

**Printer**

Independent of printer

**Reproduction steps**

Start Cura on a KDE-system with Breeze dark theme and go to settings.

**Actual results**

Parts of the settings dialog are unreadable, because the colors of background and text have way to less contrast

**Expected results**

A readable settings-dialog.

**Additional information**

This is probably the result of setting some, but not all, colors directly (by rgb-values) in the settings-GUI.

esopalumpa, 2019-08-12:
[image]
[image]

Ghostkeeper, 2019-08-14:
Duplicate of https://github.com/Ultimaker/Cura/issues/5490.

closed 2019-08-14 by Ghostkeeper