Ultimaker/Cura #6196: [4.2.1] - 680 layers jump in the gcode

Psst2, 2019-08-12:
**Application version**
<4.2.1>

**Platform**
<Windows 10 64bit>

**Printer**
<Ender 3>

**Reproduction steps**
<I sliced an STL file>

**Actual results**
<20 min in the print, the Z suddenly moved from Z0.8 to Z109, meaning it got 10 cm higher. after it reached high 109, its continued to print in the air. >

**Expected results**
< the module has 104 layers and max z high of 21mm, After I got a 10 cm height in the print,  I open the gcode file in the editor and found out that after layer 11 the next layer was 683! with Z109 instead of layer 11 with z0.8>

**Additional information**
<When I sliced again the same stl file, i didnt get this wired gap in the layers, the actual design has 104 layers. every time I am slicing I get 104 layers. I cant reproduce this gap again.  attached are the problematic gcode along a good one, both were sliced on the same module at the same project.

[CE3_T-bot_-_T-legs-1.zip](https://github.com/Ultimaker/Cura/files/3493169/CE3_T-bot_-_T-legs-1.zip)

 >

Ghostkeeper, 2019-08-14:
I can see the problem in your g-code, but I also can't reproduce it! All the layers are properly in order for me.

Ghostkeeper, 2020-04-07:
Closing this as I don't expect a reply with the additional information we requested.

closed 2020-06-19 by no-response[bot]