Ultimaker/Cura #6199: [4.2.1] I can not change nozzel. It is constant 0.4 mm

SirWinni, 2019-08-13:
**Application version**

**Platform**

**Printer**

**Reproduction steps**

**Actual results**

**Expected results**

**Additional information**

LipuFei, 2019-08-14:
@SirWinni Could you fill in the bug report form and let us know:
 - Which platform/OS you are on?
 - Which printer do you have?
 - What did you do to reproduce this issue?
 - What did you expect and what happened instead?

Ghostkeeper, 2019-08-14:
The user did fill in some of the template but didn't remove the `` so it was not displayed. I made the parts that were filled in visible.

However I'm not able to reproduce this. Here is the UM2+ with the nozzle size changed to 0.8mm:

[image]

SirWinni, 2019-08-14:
OK. they use a different route.
That's how it works for me too.
But I did it differently (see attached picture) and it does not work that way.
The value set does not change.
[image]

nallath, 2019-08-15:
This is a known bug for 4.2.1 (and has been fixed for 4.3). The workaround is to change the nozzle size through the menu bar below that.

closed 2019-08-15 by nallath

Ghostkeeper, 2019-08-15:
(See also https://github.com/Ultimaker/Cura/issues/6086).

SirWinni, 2019-08-15:
Thank you very much.