Ultimaker/Cura #6200: [4.2.1] Cura won't open or import any model

balliegojr, 2019-08-13:
**Application version**
4.2.1

**Platform**
Linux 4.19.66-1-MANJARO #1 SMP PREEMPT Fri Aug 9 18:01:53 UTC 2019 x86_64 GNU/Linux

**Printer**
Creality CR-10

**Reproduction steps**
Attempt to open any file (.stl or .jpg)

**Actual results**
Two crash report windows are shown

**Expected results**
The model should be imported without error

**Additional information**
[cura.log](https://github.com/Ultimaker/Cura/files/3499126/cura.log)
[image]

LipuFei, 2019-08-14:
Hi @balliegojr, I think there's a mismatch of the `uranium` package you have installed, because `CuraSceneNode` is a subclass of `SceneNode` from https://github.com/Ultimaker/Uranium and you can find the function `collidesWithBbox` in `SceneNode` in the 4.2 branch in Uranium.

https://github.com/Ultimaker/Uranium/blob/c64ad7bcd6b1ce28644c0f15c817745c73837f32/UM/Scene/SceneNode.py#L181-L187

I'll check the package repository for Manjaro see if there's a problem and come back to you.

LipuFei, 2019-08-14:
Hi @balliegojr , I think this is a problem with Manjaro's package repository. I checked one of the mirrors (https://mirror.neostrada.nl/manjaro/stable/community/x86_64/) and you can see that there are `cura-4.2.1-1-any.pkg.tar.xz` and `uranium-4.1.0-2.1-any.pkg.tar.xz`. Cura 4.2.1 needs uranium 4.2.0 to work.

I also checked Arch Linux's community repo and you can find uranium-4.2.0-1 there.
 - https://www.archlinux.org/packages/community/any/uranium/
 - https://www.archlinux.org/packages/community/any/cura/

Looks like Manjaro's repo is currently behind. Unfortunately, you need to wait for an update from them.

LipuFei, 2019-08-14:
I will now close this issue. You can file a bug report Manjaro about this.

closed 2019-08-14 by LipuFei