Ultimaker/Cura #6201: [4.2.1] Empty layers after scaling

kunajk, 2019-08-14:
**Application version**
4.2.1

**Platform**
Windows 10 x64, Nvidia 1060

**Printer**
Ultimaker S5 and others

**Reproduction steps**
1. Load model from attachment (stl file in T55_Wieza.zip);
2. Scale it to 120%
3. Slice

**Actual results**
Empty spaces in the middle

**Expected results**
No empty spaces

**Additional information**
This is my firs report, please calm
[image]
[T55_Wieza.zip](https://github.com/Ultimaker/Cura/files/3500068/T55_Wieza.zip)

Vandrasc, 2019-08-14:
Hi @kunajk , there were some errors in your model, after I used an STL tool repair and reloaded the model in Cura I couldn't reproduce the issue anymore, see attached the fixed file. 

You can always use the X-ray view in order to verify, in Cura, if the model is faulty or not. Faulty parts will be displayed with a red overlay. 

[objectTmp_79043451a120c297773950c33ea5489bce8f8144Suffix5d53b6cd51b1d.stl.zip](https://github.com/Ultimaker/Cura/files/3500241/objectTmp_79043451a120c297773950c33ea5489bce8f8144Suffix5d53b6cd51b1d.stl.zip)

Hope this helps, thank you for your feedback.

kunajk, 2019-08-14:
Thank you very much, sorry for troubles, have a wonderful day sir

Vandrasc, 2019-08-14:
No worries, glad I could help. FYI it is ma'am :)

closed 2019-08-14 by Vandrasc