Ultimaker/Cura #6203: Support Roof is upside down

Madpropps, 2019-08-14:
Just updated my Cura software to 4.2.1 I rely heavily on the wall for supports. AKA "enable Roof Supports" 
It's putting the roof supports on the floor of the printer and not in between the part and the support. 

**Application version**
 Cura 4.2.1

**Platform**
windows 10

**Printer**
Ender3 and Prusa i3 mk2

**Reproduction steps**

**Actual results**

**Expected results**

**Additional information**

Ghostkeeper, 2019-08-15:
I'm not able to reproduce that problem. Here is a screenshot of a model sliced for the Prusa i3, with Support and Support Roof enabled.

[image]

Question: What language are you using? Maybe it's a problem of the translation?

Otherwise, maybe you can provide a project file that reproduces the problem?

Madpropps, 2019-08-15:
I ended up re installing 4.0 and that fixed the problem.
I’ll run a test again on 4.2.1 and see if it does that again. Maybe there
was a corrupted file in the download.

closed 2020-06-19 by no-response[bot]