Ultimaker/Cura #6204: Cura 4.2.1 Ender3 Platform logo wasn't in the right location, now it's gone.

Sprint8, 2019-08-15:
**Application version**

4.2.1
**Platform**

Windows 10

**Printer**

Doesn't show at startup for specific printer

**Actual results**

**Expected results**

i avoid certain spots on the bed base on where the graphic is located.  not that it's missing, it's harder to find the sweet spot

**Additional information**

I fixed the file for the 2nd Gen Ender-3s (non-magnetic bed) for accuracy
https://mega.nz/#!M3AmwSKY!qXXWjbL3Bd6lF7TNgVkwFXcDnnD8_zIW6EyTsR6DWTY
Let me know if this is useful.

Ghostkeeper, 2019-08-15:
That's very useful. Annoying that a logo like that can significantly influence bed adhesion.

I've re-added the platform mesh that you provided. Normally I'd have given it a bit of an offset downwards to prevent Z fighting, but that prevents the logo from being seen so I'm leaving it like this.

closed 2019-08-15 by Ghostkeeper

Sprint8, 2019-08-15:
Thanks for adding this fix.  It's a simple method to use a known image point for reference when laying parts sometimes.

Liger0, 2019-08-17:
Is the bed or machine supposed to have a mesh on the ender 3 or any creality machines? I don't see anything actually.

Sprint8, 2019-08-18:
Sacha Telgenhof's settings and platform file were added since Cura 3.6 but I never complained about the bed graphic coordinates. I decided to do something about it when I installed 4.2.1 because the his Ender-3 definition file was rewritten using generic machine specs and no mention or inclusion of the bed graphic. 

I don't know why Sacha's work was deleted but so many people enjoyed it. I think it should be put back but perhaps in the rush to incorporate the Creawesome code it needed to go?  We'll see..
I only care about the Ender-3, I don't have any other Creality machines to confirm or deny correlations

Good luck!.

jonferreira, 2019-09-01:
@Sprint8 thanks, I was going mental trying to figure out what happened to the Ender logo until I finally stumbled on this

Sprint8, 2019-09-01:
> That's very useful. Annoying that a logo like that can significantly influence bed adhesion.
> 
> I've re-added the platform mesh that you provided. Normally I'd have given it a bit of an offset downwards to prevent Z fighting, but that prevents the logo from being seen so I'm leaving it like this.

It only helps because in the early weeks of ownership I've dug some divots into the bed surface and so I try to avoid those spots if I want to have a smooth finish void of any evidence of my over tramming/leveling/inexperience ;)

commit f4405355 (2019-08-15): Add platform mesh for Creality Ender-3

Provided by Sprint8 in #6204, but I think the mesh was obtained from an older Cura version.

This fixes #6204.