Ultimaker/Cura #6207: 4.1.2 Pointless moves

Arthzil, 2019-08-15:
**Application version**
4.1.2

**Platform**
Windows 10

**Printer**
Ender 3
[CE3_FanCover_LevelIndicator.zip](https://github.com/Ultimaker/Cura/files/3505857/CE3_FanCover_LevelIndicator.zip)

**Reproduction steps**
Example - layer 10, around 5th second of playing preview.

**Actual results**
 The head is making pointless zig-zag moves, no extrusion there. The moves are very rapid and are creating artifacts and stringing.

**Expected results**
The move shouldn't even happen.

**Additional information**

smartavionics, 2019-08-15:
Turning off the Avoid Printed Parts When Travelling reduces the travels somewhat.

Ghostkeeper, 2019-08-16:
Indeed this is a bug. It should shortcut the combing move there as long as there is nothing in the way. We'll investigate.

dferrier1, 2020-02-21:
fyi...

I am seeing this same behavior in version 4.4.1 running windows 8.1

dferrier1, 2020-02-21:
Version 4.4.1 running windows 8.1.

In my gcode file, the behavior starts immediately from the second layer on. 

[CCR10MAX_CR 10 S5 four Trays With Divider v3.zip](https://github.com/Ultimaker/Cura/files/4236816/CCR10MAX_CR.10.S5.four.Trays.With.Divider.v3.zip)

smartavionics, 2020-02-21:
Please save the project file and then either rename the .3mf file to zip or zip it and attach to this issue. Without that it's hard to help. Thanks.

dferrier1, 2020-02-21:
My apologies.

Here is the .3mf file. I did not save the original as .3mf so this one is actually simpler but still shows the same behavior. I noticed that I had selected a different profile and it did *not* show the behavior. But with this profile selected (the same as the original) I do see the behavior. I have scoured the settings but do not see anything that could explain this though. 

[CCR10MAX_CR 10 S5 Tray With Divider v3.zip](https://github.com/Ultimaker/Cura/files/4236915/CCR10MAX_CR.10.S5.Tray.With.Divider.v3.zip)

smartavionics, 2020-02-21:
You have the wipe nozzle thing enabled, turn that off and the wacky moves go away.

[image]

dferrier1, 2020-02-22:
Ah yes, I just found that. I had to enable all options visibility to see it though. 

Thanks.

Ghostkeeper, 2020-02-27:
The original issue is still on our backlog (developers, ticket number is CURA-6725).

OutsourcedGuru, 2020-04-08:
For what it's worth, I was just scratching my head over this one. There are many sections (large and small) of seemingly-pointless `G0` moves. The sheer number makes me think that this must be a bug. I've tried different infill types and percentages.

Versions: **Cura 4.5.0** and 3.6 on macOS 64-bit
Wipe Nozzle Between Layers: OFF

[5milCamRing_Gyr45.gcode.zip](https://github.com/Ultimaker/Cura/files/4452755/5milCamRing_Gyr45.gcode.zip)

Ghostkeeper, 2020-04-14:
In your case the travel moves that I consider "pointless" are most likely solved by setting the Z Seam Position to "Shortest".

OutsourcedGuru, 2020-04-14:
I changed to Shortest as seen here.
[image]

Nine or ten `G0`s in a row:
```
G1 X54.356 Y54.203 E0.02864
G0 F1800 X57.277 Y52.44
G0 X58.79 Y51.714
G0 X60.386 Y51.197
G0 X62.036 Y50.893
G0 X63.709 Y50.811
G0 X65.38 Y50.949
G0 X67.02 Y51.307
G0 X68.598 Y51.878
G0 X69.803 Y52.504
G0 X73.274 Y54.607
G1 F600 X75.612 Y56.944 E0.10995
```

Is there some reason for this? Is it trying to wipe off excess extrusion in some way?

Ghostkeeper, 2020-04-21:
It makes multiple travel moves after one another usually because it's trying to avoid hitting a wall, i.e. combing. It will also travel to the inside if it was on a wall while the travel move happens, and will travel through the inside of the object to the closest place of another object if it travels from one object to another. See [this documentation](https://github.com/Ultimaker/CuraEngine/blob/master/docs/generating_paths.md#travelling) for details.

There are also more cases where it might make multiple travel moves after one another, e.g. for an extruder switch with wipe or wiping off the prime tower, but those are pretty easy to recognise in the g-code. Yours just seems to be moving towards the right somewhere while avoiding something with a slightly round edge in the middle.

Ghostkeeper, 2021-06-17:
This issue was taken off of our backlog, when the quality engineers reassessed priorities. Because the travel moves are usually a tiny fraction of the print time and this issue is really uncommon, we'll have to defer this. That means we're not expecting to fix this issue any time soon (unless more critical cases are discovered).

closed 2023-06-24 by github-actions[bot]