Ultimaker/Cura #6208: [4.2.1] Crash on Startup

UltJan, 2019-08-15:
**Application version**
3.6.0, 4.0.0, 4.2.1

**Platform**
Windows 10 - Version 10.0.18362.239
Ryzen 7 3700X
RTX 2070 Super

**Printer**
i3, i3 Mega, Black Widow

**Reproduction steps**
install and start cura

**Actual results**
crash after the loading screen

**Expected results**
cura should start

**Additional information**
Steps i tried:
- clean install cura 3.6.0, 4.0.0 and 4.2.1
- deleted files and folders in APPDATA/Roaming and /Local
- checked OpenGL
- checked NVIDIA drivers
- unplugged other displays and only used 1080p
- checked the Error log, can't help myself

I searched GitHub and Google and can not find a way to work around.
Untill today it worked fine. Now it won't start.

**Error-Log**
[stderr.log](https://github.com/Ultimaker/Cura/files/3506331/stderr.log)

Ghostkeeper, 2019-08-16:
The log file seems to indicate that Qt crashed while loading the main window? Very strange. There is not a lot to go on, as you've already said.

rburema, 2019-08-16:
I searched, there was a similar issue a while back on the forum, it turned out that in that case, Cura tried to use the intel integrated graphics card instead of the NVIDIA one:

https://community.ultimaker.com/topic/27801-solved-cura-was-fine-but-now-wont-open-without-external-monitor/

That crash specifically mentions OpenGL though, this one doesn't.

UltJan, 2019-08-16:
Thank you for your help.
Today it started without hesitation.

I will close this now.
I can not reproduce the error.

closed 2019-08-16 by UltJan

Ghostkeeper, 2019-08-19:
It may be related to where Cura is located on the desktop. Which monitor is displaying Cura can determine which graphics card is used.