Ultimaker/Cura #6209: Separe the support zhop setting

Liger0, 2019-08-15:
**Is your feature request related to a problem? Please describe.**
Not a printing issue but a profile management. The zhop works differently if support is enabled or not and it creates confusion. For example it enables or disables itself differently depending on the support presence. I didn't notice this until not just some days ago.

**Describe the solution you'd like**
Put instead a zhop sub-option that forces it to be enabled or not if supports are enabled.

**Describe alternatives you've considered**
Set an option like "zhop on support" in the support section instead for more visibility.

**Affected users and/or printers**
Any?

rburema, 2019-08-16:
I don't think z-hops should be influenced by support. What does your "Z Hop Only Over Printed Parts" say?

Ghostkeeper, 2019-08-16:
It is influenced by support due to this line here:

https://github.com/Ultimaker/Cura/blob/60758093f1ac7511f4f72daeb1c13fc235b8257e/resources/definitions/creality_base.def.json#L93

Liger0, 2019-08-16:
But it is very confusive. Like, if I have support enabled, and I enable zhop, then zhop is enabled on supports but still disabled on the rest of the print. In fact, disabling the supports, the zhop results disab led again.

theWaldschrat, 2019-08-19:
I like to second that it's confusing, even more so that it didn't behave that way before 4.2.
Apparently there were some changes to the z-hop mechanisms, because there is now an option "Z Hop Only Over Printed Parts" which hadn't been there before. So I guess it was introduced with that changes.
But anyway, I don't think above argument of it being an option that could be in multiple section holds water: I'd be rather surprised if anyone excepts Z-hop in the support section. Or it to have anything to do with support at all.
And I'd also say, if you'd ask your user base, close to 100% would be confused by Z-hop being enabled when you tick on support.

"Just switch it off" is also only somewhat helpful, because it'll be switched on for any new profile.

Ghostkeeper, 2019-08-21:
> Apparently there were some changes to the z-hop mechanisms, because there is now an option "Z Hop Only Over Printed Parts" which hadn't been there before.

This setting was introduced in October 2016; I think that was with the 2.3 release. We did change the Maximum Z Feedrate Override setting to the Z Hop Speed setting and made Z hops the only thing that it affects. That's the only change we made to Z hops in 4.2.

When we get submitted a profile that says to enable Z hops when supports are active, we'd assume that supports tend to get knocked over by blob forming and then Z hops prevent that issue. There's nothing we can test in practice in this so who am I to say "no, your profiles are bad"?

theWaldschrat, 2019-08-22:
Oh, right, that option was hidden by some other options in 4.1. Nevermind then.

Once I had a layer shift that I put down to the nozzle hitting printed parts, but there it hit infill and not support. But I didn't catch the printer doing it, so it may have been caused by something else.
Much more often I had been fighting stringing and enabling Z-hop greatly increases stringing on bowden setups like the Creality machines. Not only in my personal experience but also for example in this video:
https://www.youtube.com/watch?v=_QRb54zVPfQ
So for me, Z-hop is off by default...well, always, actually, because as I said, I never had something knocked over.
Anyway, I'd say it might be better to have Z-hop disabled by default to reduce stringing and leave it to the user to switch it on when she needs it. And not having it randomly change on her when she clicks some seemingly unrelated option.

Also I'm not saying you should have caught that before release, because I know that you have no way to test it. But I'm saying you should change it in the next release because it would improve things. Or at least cause less confusion.

Liger0, 2019-08-22:
I also can't use zhop because of huge stringing. Also the absence of zhop doesn't cause any support knock off.
I'd suggest to put a "enable zhop only if supports are enabled" checkbox for who may desire it.

closed 2024-12-09 by github-actions[bot]