Ultimaker/Cura #6210: Print Direction Reversing

MaidDeLaMer, 2019-08-16:
**Application version**
4.2.1

**Platform**
Windows 10 Enterprise, Version 1803. OS Build 17134.885.

**Printer**
Ultimaker-003aad, Ultimaker 3 Extended 
[UM3E_coin.3mf.zip](https://github.com/Ultimaker/Cura/files/3510113/UM3E_coin.3mf.zip)

**Reproduction steps**
Allowed Cura to automatically slice the coins, and left all settings at default. Sent to printer and almost immediately began encountering issues.

**Actual results**
Extruder changes direction and ruins print.

**Expected results**
Extruder should continue in the same directions so as not to pull up the freshly laid filament.

**Additional information**
Extruder will print one ring of the coin in a clockwise direction and then immediately reverse directions to print the second ring, catching on the freshly laid filament and pulling it up off the print bed. Adding a brim has no effect, increasing the infill % makes it worse, changing the layer height has little effect.

smartavionics, 2019-08-16:
Cura always prints holes in one direction and the outside of the part in the other direction. Possible setting changes to reduce the chance of ripping up the printed lines are changing the z seam alignment to "user specified" and also setting the minimum wall flow to a value such as 50. Hope this helps.

Ghostkeeper, 2019-09-16:
We don't consider this a bug per sé as it's expected behaviour right now. The reason this is happening is because the contours are stored like this in ClipperLib. It's a convention in computer science that positive polygons should be stored in counter-clockwise order of the vertices, while negative polygons should be stored in clockwise order. CuraEngine just follows this convention.

We have plans to change that. Holes inside polygons would then be printed in counter-clockwise order as well.

However this won't change it for all shapes. For instance, if you were to have a C shape, like a ring but with a small opening, the direction of the inner side will still be clockwise rather than counter-clockwise. That's a limitation of geometry.

BagelOrb, 2019-10-29:
You would want the same slicing directions if one layer is a C shape and the next layer is an O shape.

Ghostkeeper, 2020-11-18:
Closing this as a duplicate of https://github.com/Ultimaker/Cura/issues/1867.

dwery, 2022-03-06:
@Ghostkeeper any chance this will ever be fixed or user-selectable?

Ghostkeeper, 2022-03-20:
Yes, in fact we have a ticket on our backlog about this. Developers, see issue CURA-6230. I'll re-open this issue since it's still an issue.

In Cura 5.0 we've added a feature to make the walls alternate at every inset, on purpose. This reduces warping. With 5.0 this sort of stuff has been made easier to implement. We had to implement a way of inverting the direction of walls, because it can create center-walls that are not closed loops, and you want to be able to print those in the other direction to optimise the travel moves (to start printing from the endpoint that is closest). So having implemented that, purposefully reversing certain walls is easier.

adripo, 2023-01-11:
Any update on this issue?

jtiscione, 2023-06-12:
The new dual-color and tri-color filaments that are coming out really bring out this issue.
[image]
I also tried getting rid of that ugly 12 o clock seam using the "random" setting for Z-seam alignment, but if I do that, the placement of counterclockwise and counterclockwise paths is completely random.

dragorosson, 2023-06-28:
Cura 5.2.2. I am getting similar to this as well. My part is cylindrical but not uniform top to bottom. The thinnest walls are just 2 lines (so, both lines are "outer" walls in Cura?). Layers in regions 2 and 3 are identical (model-wise), yet Cura prints the inside wall first in one and second in the other, resulting in a slightly different surface finish. It also puts the z-seam start and stop in slightly different locations even with the "user specified" option.

P.S. I see the same behavior on `5.4.0-beta1` except the nozzle always(?) goes CCW on the outermost layer.

[image]

- Region 2: Outermost wall printed 2nd
- Region 3: Outermost wall printed 1st
[image]

idmadj, 2023-08-08:
@dragorosson I have also experienced issues with these behaviors.

To summarize, there are three inconsistency problems across layers:
1. **Walls CW/CCW direction.** An option to force a direction would be very helpful to get uniform-looking walls;
2. **Hole vs perimeter order.** When a feature is only composed of one inside wall and one outside wall, the order in which they are printed varies across layers, resulting in surface finish defects. Cura seems to already differentiate between holes and perimeters, so it might be possible to ensure that hole walls are printed before perimeter walls, or the opposite, on every layer.
3. **Z-seam position.** The latest releases have improved on that but it could be improved further;