Ultimaker/Cura #6211: [4.2.1] Profile dropdown disappears

Liger0, 2019-08-16:
**Application version**
4.2.1

**Platform**
Win10 x64

**Printer**
Cr-10S

**Reproduction steps**
It happens randomly. PC restart usually fix it.

**Actual results**
Sometimes the big menus, when closed, disappear and go up of the list, sometimes you can see it half on the very top margin, sometimes not at all. 

**Expected results**
...

**Additional information**
https://youtu.be/jbSGiBmMSSI

Ghostkeeper, 2019-08-19:
I could reproduce it as well after a couple of minutes of clicking and scrolling. It seems to be a render bug or maybe something with the ScrollView in there. Here are reproduce steps that seem to reproduce the problem 100% of the time:

1. Show all settings.
2. Make sure all setting categories are collapsed.
3. Expand the Shell category.
4. Scroll down so that the Shell category header is out of view.
5. Scroll back up again.
6. Collapse the Shell category header.

It's always that Shell category header that does it. I blame Big Oil.

LipuFei, 2019-08-20:
JIRA ticket: CURA-6728

fatalelieberi, 2019-11-27:
Hi, I'm having the same problem with Cura 4.4 too although for me reproducing it is much simpler:
1. Open the settings in Expert View
2. Expand the Shell category
3. Collapse the Shell category without scrolling at all
4. All profile dropdowns disappear

To fix it, just open other profile view other than Expert view.

[Cura 4.4 Interface Bug Video.zip](https://github.com/Ultimaker/Cura/files/3894650/Cura.4.4.Interface.Bug.Video.zip)

closed 2024-12-07 by github-actions[bot]