Ultimaker/Cura #6212: download fail？

robin1015, 2019-08-17:
To upgrade and install the latest version of cura（window7 64）, the download of installation files always fails. I have tried for several days. Version: Ultimaker Cura 4.2.1, etc. Address: Shanghai, China.

LipuFei, 2019-08-19:
Hi @robin1015 , our download is now on GitHub, which is hosted on AWS US. It can be blocked by "the firewall". Could you checkout Ultimaker's CN site see if that works for you?

http://ultimaker.cn/Ultimaker-cura-software/Download

LipuFei, 2019-08-19:
BTW, I asked some of my friends in Shanghai to test the links on both GitHub and ultimaker.cn. Both worked for them. Could you try it again?

Ghostkeeper, 2019-08-19:
Ultimaker's website has a provider specific to China to prevent the issue of the "Great" Firewall. Download from ultimaker.com rather than from Github, I'd advise.

robin1015, 2019-08-20:
thank very much。https://ultimaker.com/en/products/ultimaker-cura-software，，The website I used in the past is not available in China. The website you gave is OK. Thank you very much.

robin1015, 2019-08-20:
ultimaker.cn ok

robin1015, 2019-08-20:
> Ultimaker's website has a provider specific to China to prevent the issue of the "Great" Firewall. Download from ultimaker.com rather than from Github, I'd advise.

http://ultimaker.cn/Ultimaker-cura-software/Download  ok            
thank you

closed 2019-08-20 by LipuFei