Ultimaker/Cura #6213: [4.2.1] "Add printer by IP address" (or hostname) dialog is limited to hexadecimal character set

JackBeQuick87, 2019-08-17:
**Application Version**

4.2.1

**Platform**

Windows.  (Others not untested).

**Steps to Reproduce**

Go to Settings>"Add Printer">"Add printer by IP".
Attempt to add a printer using a hostname containing characters outsize of the hexadecimal [0-9, Aa-Ff] subset.

**Actual Results**

The input field is limited to characters {0-9, .(decimal), a-f, A-F}.

**Expected results**

Users should be able to enter a hostname in that input box.  For example: "myprinter.mynetwork.lan"

Ghostkeeper, 2019-08-19:
It doesn't do DNS resolution at the moment. It's not just the limitation of the input field you're asking from us!

fieldOfView, 2019-08-19:
The UI says "Add the IP address _or hostname_ of your printer on the network" (my emphasis).

Ghostkeeper, 2019-08-19:
Oh, you're right. The button says "Add by IP", the window header says "Add printer by IP address" and the textbox says "Please enter your printer's IP address", but the label for the text box indeed says "Enter the IP address or hostname of your printer on the network".

LipuFei, 2019-08-20:
If we allow adding by hostname, you will be able to enter any random string, and those cases are not handled very well at the backend right now... We discussed this and decided to defer it for now.

Ghostkeeper, 2019-08-21:
Since it is deferred and not fixed yet, we'll keep this issue open in case someone wants to take a look at it.

I'll at least change that string that says you can enter hostnames there though.

SteveDC2, 2019-09-07:
Until this is truly fixed I would suggest removing  "or hostname" from the description. I thought I was going crazy trying to enter my host name and took me a while to work out not all characters were being processed!!!

Ghostkeeper, 2019-09-09:
Yeah, I did: https://github.com/Ultimaker/Cura/commit/9122d0cca44d684d937898afa6aa7b39e5180d02

fieldOfView, 2020-03-23:
This monstrosity does a good job at validating hostnames and ip addresses:
`^(([0-9]|[1-9][0-9]|1[0-9]{2}|2[0-4][0-9]|25[0-5])\.){3}([0-9]|[1-9][0-9]|1[0-9]{2}|2[0-4][0-9]|25[0-5])$|^(([a-zA-Z]|[a-zA-Z][a-zA-Z0-9\-]*[a-zA-Z0-9])\.)*([A-Za-z]|[A-Za-z][A-Za-z0-9\-]*[A-Za-z0-9])$|^(?:(?:(?:(?:(?:(?:(?:[0-9a-fA-F]{1,4})):){6})(?:(?:(?:(?:(?:[0-9a-fA-F]{1,4})):(?:(?:[0-9a-fA-F]{1,4})))|(?:(?:(?:(?:(?:25[0-5]|(?:[1-9]|1[0-9]|2[0-4])?[0-9]))\.){3}(?:(?:25[0-5]|(?:[1-9]|1[0-9]|2[0-4])?[0-9])))))))|(?:(?:::(?:(?:(?:[0-9a-fA-F]{1,4})):){5})(?:(?:(?:(?:(?:[0-9a-fA-F]{1,4})):(?:(?:[0-9a-fA-F]{1,4})))|(?:(?:(?:(?:(?:25[0-5]|(?:[1-9]|1[0-9]|2[0-4])?[0-9]))\.){3}(?:(?:25[0-5]|(?:[1-9]|1[0-9]|2[0-4])?[0-9])))))))|(?:(?:(?:(?:(?:[0-9a-fA-F]{1,4})))?::(?:(?:(?:[0-9a-fA-F]{1,4})):){4})(?:(?:(?:(?:(?:[0-9a-fA-F]{1,4})):(?:(?:[0-9a-fA-F]{1,4})))|(?:(?:(?:(?:(?:25[0-5]|(?:[1-9]|1[0-9]|2[0-4])?[0-9]))\.){3}(?:(?:25[0-5]|(?:[1-9]|1[0-9]|2[0-4])?[0-9])))))))|(?:(?:(?:(?:(?:(?:[0-9a-fA-F]{1,4})):){0,1}(?:(?:[0-9a-fA-F]{1,4})))?::(?:(?:(?:[0-9a-fA-F]{1,4})):){3})(?:(?:(?:(?:(?:[0-9a-fA-F]{1,4})):(?:(?:[0-9a-fA-F]{1,4})))|(?:(?:(?:(?:(?:25[0-5]|(?:[1-9]|1[0-9]|2[0-4])?[0-9]))\.){3}(?:(?:25[0-5]|(?:[1-9]|1[0-9]|2[0-4])?[0-9])))))))|(?:(?:(?:(?:(?:(?:[0-9a-fA-F]{1,4})):){0,2}(?:(?:[0-9a-fA-F]{1,4})))?::(?:(?:(?:[0-9a-fA-F]{1,4})):){2})(?:(?:(?:(?:(?:[0-9a-fA-F]{1,4})):(?:(?:[0-9a-fA-F]{1,4})))|(?:(?:(?:(?:(?:25[0-5]|(?:[1-9]|1[0-9]|2[0-4])?[0-9]))\.){3}(?:(?:25[0-5]|(?:[1-9]|1[0-9]|2[0-4])?[0-9])))))))|(?:(?:(?:(?:(?:(?:[0-9a-fA-F]{1,4})):){0,3}(?:(?:[0-9a-fA-F]{1,4})))?::(?:(?:[0-9a-fA-F]{1,4})):)(?:(?:(?:(?:(?:[0-9a-fA-F]{1,4})):(?:(?:[0-9a-fA-F]{1,4})))|(?:(?:(?:(?:(?:25[0-5]|(?:[1-9]|1[0-9]|2[0-4])?[0-9]))\.){3}(?:(?:25[0-5]|(?:[1-9]|1[0-9]|2[0-4])?[0-9])))))))|(?:(?:(?:(?:(?:(?:[0-9a-fA-F]{1,4})):){0,4}(?:(?:[0-9a-fA-F]{1,4})))?::)(?:(?:(?:(?:(?:[0-9a-fA-F]{1,4})):(?:(?:[0-9a-fA-F]{1,4})))|(?:(?:(?:(?:(?:25[0-5]|(?:[1-9]|1[0-9]|2[0-4])?[0-9]))\.){3}(?:(?:25[0-5]|(?:[1-9]|1[0-9]|2[0-4])?[0-9])))))))|(?:(?:(?:(?:(?:(?:[0-9a-fA-F]{1,4})):){0,5}(?:(?:[0-9a-fA-F]{1,4})))?::)(?:(?:[0-9a-fA-F]{1,4})))|(?:(?:(?:(?:(?:(?:[0-9a-fA-F]{1,4})):){0,6}(?:(?:[0-9a-fA-F]{1,4})))?::))))$`

http://jsfiddle.net/AJEzQ/

Ghostkeeper, 2020-03-24:
Hehe. Well, Qt only allows typing if after the keystroke it is matched by the regex. So for Qt we'd need to have it accept parts of IPs or hostnames as well as you type! Or we'd need to validate upon clicking "next", which is maybe a better idea anyway.

LANversations, 2022-05-05:
This issue is still present in 5.0.0 beta. Can this be move from deferred to active issue or anything?

closed 2023-07-13 by github-actions[bot]