Ultimaker/Cura #6214: [4.2.1] Add a printer - printer list fails to populate correctly

b4545, 2019-08-17:
**Application version**
4.2.1

**Platform**
Win x64

**Reproduction steps**
New install of 4.2.1 on freshly formatted computer. Go to add a new printer and view list of printers. _Do not immediately scroll down. This makes the issue disappear._

**Actual results**
Some radio buttons and printer names are not displayed correctly

**Expected results**
All printer manufactures should have a full list of printers displayed

**Additional information**
Scrolling to the bottom of the list and then back up appears to solve the issue

[image]
[image]

Ghostkeeper, 2019-08-19:
We saw this as well when developing this UI for the 4.1 release. We have no idea what causes this but the QML properties seem to be in order. It just doesn't seem to update the render after the expansion. Possibly this is a multi-threading issue in Qt. It's slightly out of our power to fix this for now. We'll keep an eye on it though.

nallath, 2020-06-22:
We have found a workaround for this, which has been implemented since.

closed 2020-06-22 by nallath