Ultimaker/Cura #6215: [4.2.1 & master] Skin region inverted when simplify() applied

smartavionics, 2019-08-17:
**Application version**

**Platform**

Linux

**Printer**

ANET A8 & Kossel XL

**Reproduction steps**

Sliced attached project with no changes.

**Actual results**

Some skin regions become inverted (hole where skin should be and skin where hole should be).

**Expected results**

Perfection, of course!

**Additional information**

Here is what you get...

[image]

And it should look like this...

[image]

A workaround is to set the top/bottom line width to a slightly different value, i.e. 3.9 or 4.1 instead of 4.0.

Delving into the engine, commenting out the call to simplify() in generateSkinInsets() also fixes the issue:
```
void SkinInfillAreaComputation::generateSkinInsets(SkinPart& skin_part)
{
    for (size_t inset_idx = 0; inset_idx < skin_inset_count; inset_idx++)
    {
        skin_part.insets.push_back(Polygons());
        if (inset_idx == 0)
        {
            //The 10 micron reduced inset is to prevent rounding errors from creating gaps that get filled by the fill small gaps routine.
            skin_part.insets[0] = skin_part.outline.offset(-skin_line_width / 2 + 10);
        }
        else
        {
            skin_part.insets[inset_idx] = skin_part.insets[inset_idx - 1].offset(-skin_line_width);
        }

        // optimize polygons: remove unnecessary verts
        skin_part.insets[inset_idx].simplify();   <--- comment this out to stop skin being inverted
        if (skin_part.insets[inset_idx].size() < 1)
        {
            skin_part.insets.pop_back();
            return; // don't generate inner_infill areas if the innermost inset was too small
        }
    }
}
```

[led_panel.3mf-renamed.zip](https://github.com/Ultimaker/Cura/files/3512010/led_panel.3mf-renamed.zip)

smartavionics, 2019-08-17:
This bug also occurs on the armhf build so it's not specific to the x86_64 FPU.

smartavionics, 2019-08-17:
Did a little bit of investigation. When it goes wrong, skin_part.insets[inset_idx] after the call to simplify() appears to have lost a polygon. i.e. before the call, the size is 2 and after the call the size is 1 and the resulting polygon area is negative.

Ghostkeeper, 2019-08-19:
Strange. The hole seems to be large enough that simplify shouldn't affect it, right?

smartavionics, 2019-08-19:
I think it's losing the outer triangle shaped outline and leaving the hole.

rburema, 2019-08-20:
While I see a bug in the model (tested with 4.2.1 and current master, on Windows), it doesn't behave the same for me as for you, it seems:

[image]

The hole is still erronously filled, but the proper skin is 'back'. Unfortunately, commenting out the simplify call doesn't work in this case (it gives the same issue as in the screenshot above).

smartavionics, 2019-08-20:
Interesting that in the smaller image the hole is filled at a lower layer than shown in the big image.

rburema, 2019-08-20:
Yes, strange. The inset image is from one of the other corners of the model. There, the hole-fill is made at the bottom of the top skin, instead of at the top of the top skin.

rburema, 2019-08-20:
(Devs: ticket CURA-6729 was made for this issue.)

smartavionics, 2019-08-20:
I note that if this test in simplify() is disabled, the error doesn't occur:

```
        const coord_t base_length_2 = vSize2(next - previous);
        if (base_length_2 == 0) //Two line segments form a line back and forth with no area.
        {
            continue; //Remove the vertex.
        }
```

What's happening is that the original polygon(s) have 4 or 5 points but after simplification with this code in place, they have less than 3 points and so get dropped.

smartavionics, 2019-08-20:
I have a fix that I am testing and will submit a PR soon if it looks OK.

smartavionics, 2020-07-14:
Should now be fixed. Closing...

closed 2020-07-14 by smartavionics