Ultimaker/Cura #6216: 4.2.1 Unnecessary Moves Between Layers

timodreynolds, 2019-08-17:
When using 4.2.1 I get an unnecessary move to the same spot (far away from the item(s) i'm printing) at the end of one layer before the next layer.  When using 4.1 I don't have that problem.  I'll update with a youtube video in a bit.

smartavionics, 2019-08-17:
What would be much more useful would be if you could save the project file (File -> Save), zip or rename the .3mf to .zip and attach that to this issue. Thanks.

smartavionics, 2019-08-17:
Also, a copy of the gcode would be useful. Thanks.

timodreynolds, 2019-08-17:
Here's the GCODE for 4.2.1 vs. 4.1

[CCR10S_1-8th NPT Filament Guide v4 - Cura 4.2.1.txt](https://github.com/Ultimaker/Cura/files/3512082/CCR10S_1-8th.NPT.Filament.Guide.v4.-.Cura.4.2.1.txt)
[CCR10S_1-8th NPT Filament Guide v4 - Cura 4.1.txt](https://github.com/Ultimaker/Cura/files/3512083/CCR10S_1-8th.NPT.Filament.Guide.v4.-.Cura.4.1.txt)

timodreynolds, 2019-08-17:
Here's a picture of Cura 4.2.1 telling me it's going to do this move at the end of the layer...
[image]

timodreynolds, 2019-08-17:
[Pre-sliced file CCR10S_1-8th NPT Filament Guide v4 - Cura 4.2.1.zip](https://github.com/Ultimaker/Cura/files/3512094/Pre-sliced.file.CCR10S_1-8th.NPT.Filament.Guide.v4.-.Cura.4.2.1.zip)

smartavionics, 2019-08-17:
There's no model in that .3mf file so I can't slice it! Can you provide the model, please?

timodreynolds, 2019-08-17:
had to put the .STL in a zip
[1-8th NPT Filament Guide v4 3-5.25.STL.zip](https://github.com/Ultimaker/Cura/files/3512119/1-8th.NPT.Filament.Guide.v4.3-5.25.STL.zip)

smartavionics, 2019-08-17:
Thanks for the files. I think the travels you are seeing are because you have the Lift Head setting enabled and the layer times are very short. If you disable that, you don't get the travels...

[image]

timodreynolds, 2019-08-17:
Something related to that changed in 4.2.1?

smartavionics, 2019-08-17:
Don't think so, it looks like it has just been set in the profile. By default, it's no enabled.

timodreynolds, 2019-08-18:
OK, but lift head was enabled on 4.1 and doesn't do the same thing.

timodreynolds, 2019-08-18:
I looked at the feature on Cura and yes it seems that it's doing exactly what it says it will.  So I guess that this "feature" wasn't working correctly on 4.1 which was fortunate for me.  I'm not sure how this is of use, however.  For my prints, it just gives the material time to extrude out and become extra junk built up in the print.

Ghostkeeper, 2019-08-19:
> I'm not sure how this is of use, however.

The purpose of this is to let the previous layer cool down and solidify before placing the next layer on top. This prevents sagging and an uneven surface. See also https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/cool_lift_head.md
Of course, every setting has disadvantages as well. In this case it's that you get oozing if you have materials that are hard to retract, or materials that are particularly runny.

This can be prevented by reducing the Minimum Speed, reducing the Minimum Layer Time or disabling Lift Head. The location where it's moving can't be configured with a setting right now. (And it's one of the known cases where the nozzle can move outside of the build volume...)

I'm closing this as it's working as intended. I don't recall any bug fix that we made in this area for the 4.2 release though so I don't know why it wasn't working for you in 4.1.

closed 2019-08-19 by Ghostkeeper