Ultimaker/Cura #6217: [4.2.1] Settings visibility - information icons not updating correctly

b4545, 2019-08-17:
**Application version**
4.2.1

**Platform**
Win x64

**Reproduction steps**
I noticed top surface flow setting wasn't visible, so I went to visibility settings and saw the information icon that said the top skin layers was set to 0. I updated top skin layers to 1, and the surface flow setting was made visible again. However, in the settings visibility window, the information icon is still displayed and it still says that the setting is still hidden, when it is not.

**Actual results**
Display windows says that settings are hidden, when they are in fact not hidden

**Expected results**
Settings should only say "hidden" when they are truly hidden

**Additional information**
Issue not resolved after restarting Cura

Minor issue, just trying to document little things when I see them.
[image]

Ghostkeeper, 2019-08-19:
Yup. I can reproduce this.

Maybe it's got something to do with global vs. per-extruder settings.

Ghostkeeper, 2019-09-16:
We've decided to defer this issue. It's not a big problem and we don't have time to fix this soon. But you have a good eye, it's a hard bug to find. We'll keep it open in case someone encounters this again.

printingotb, 2020-11-02:
I noticed this as well, in 4.8 beta. (And I have actually been told my eyes are too good for quality control and similar things :sweat_smile: )
But as you say, not an important bug.
[image]

printingotb, 2023-05-30:
Can confirm that the bug is still present in 5.3.1

MariMakes, 2023-05-30:
Hey @printingotb,

Fresh set of eyes reporting for duty 👍 
Thanks for confirming this in the latest version of Cura 😊 

I'll bring it up to the team to see what they can do to improve it. 
But I expect that if it's not a quick fix that it will not get a high priority to implement. 
Fingers crossed 🤞

MariMakes, 2023-06-01:
Hey @printingotb,

We discussed your issue and some of us are concerned that this points to something else not really working so we've added a Technical improvement ticket to the backlog with the intent to improve this. 
For internal reference CURA-10645

Thanks for the report! 👍

closed 2024-12-07 by github-actions[bot]