Ultimaker/Cura #6218: 0.6 (and other not predefined) nozzle size not possible in 4.2.1?

pkoretic, 2019-08-17:
**Application version**
4.2.1

**Platform**
x86_64 GNU/Linux

**Printer**
Ender 3

**Reproduction steps**
[image]

**Actual results**
0.6 nozzle size is not available

**Expected results**
0.6 nozzle size is available

**Additional information**
With 4.2.1 I cannot enter anymore specific nozzle size, 
and I see two similar bugs

https://github.com/Ultimaker/Cura/issues/6199
https://github.com/Ultimaker/Cura/issues/6086

but I don't see how I could add manually new size of the nozzle now - since I have a bunch of different nozzle sizes not available in the menu and the setting for manual addition is not available anymore in the printer settings?
[image]

Maybe I'm missing something obvious so thanks in advance.

pkoretic, 2019-08-18:
Ok, interesting, after I restarted cura I have 0.6 in my list, but still would need to add other sizes I have that are not on the list.

LipuFei, 2019-08-19:
@pkoretic The problem that 0.6mm is missing from the list indeed seems like a bug. Do you remember what you did when you saw the bug?

Currently in Cura 4.2.1, Creality Ender 3 has 0.2, 0.3, 0.4, 0.5, 0.6, 0.8, and 1.0 as available options.

Ghostkeeper, 2019-08-19:
I'm not able to reproduce this bug. It's showing 0.6mm in the list.

Workaround is to change the line width to 0.66mm, since that seems to be the only real change in this profile. I'm saying "real" because I'm not counting the Raft Base Line Width which is configured separately from the nozzle size.

pkoretic, 2019-08-20:
I've added it as a new printer, I can try to reproduce it again.
But how would I add size 1.2 or 0.7?

Ghostkeeper, 2019-08-21:
To add it properly, you'd add a file for `creality_base_1.2.inst.cfg` to your resources folder in the `variants` subfolder. You can look at this file for an example: https://github.com/Ultimaker/Cura/blob/master/resources/variants/creality_cr10_1.0.inst.cfg In this profile you would add all of the settings necessary to make it print well with a 1.2mm nozzle. Keep in mind though that the material profiles and quality profiles will override the changes you make in the variant (nozzle) profile.

Realistically, to make Cura push more material through your nozzle to account for a bigger nozzle orfrice, you simply increase the line width to 1.2mm or whatever works best.

pkoretic, 2019-08-31:
Well, I got it again, just after restarting cura few times. If it means anything I have my printer connected over serial.

[image]

It got fixed by restarting Cura again.

closed 2020-06-19 by no-response[bot]