Ultimaker/Cura #6219: [4.2.1] Missing mesh around creality printers?

Liger0, 2019-08-17:
**Application version**
4.2.1

**Platform**
Win10 xs64

**Printer**
Any creality

**Reproduction steps**
None

**Actual results**
I don't recall ever seeing a 3d model around the build plate on any creality machines not on the build plate.

**Expected results**
I did just read the creawsome implementation was supposed to add it, but I never saw it. I asked some people and it seems some have it, some don't just like me.

**Additional information**

Ghostkeeper, 2019-08-19:
The original CreawsomeMod had meshes for [Ender 3, CR10s Pro and CR10s](https://github.com/trouch/CreawsomeMod/tree/master/src/meshes).

The meshes for the Ender 3 and CR10s are too high resolution to add to Cura's mainstream release. To keep the download size manageable and for performance of rendering we have a guideline limit of 1MB per platform mesh.

The model for the CR10s Pro could still be added though. It wasn't included in the PR that merged it in.

Liger0, 2019-08-19:
So it's normal not having any meshes on the cr10s and ender 3?

Ghostkeeper, 2019-08-19:
As long as nobody submits any model, yes. We've just had a model submitted for the Ender 3 though so the next release will have one there.

closed 2019-08-19 by Liger0