Ultimaker/Cura #6220: Permission denied ERRNO13.

fosterswat, 2019-08-18:
When i try to save on sd card it fail and say permission denied errno 13 can't save the file. i can save it on pc but it become protected and can't be transfered. the only way i have to print is with usb. Did someone know what to do to be able to save again. It's the first days that make me this message from the last 6 month it is really annoiying. Thanks for your help and all you're advices. 

**Application version**
4.2.1

**Platform**
windows 10

**Printer**
Ender 3

**Reproduction steps**
every .stl file i have take on thingiverse or my own design are impossible to save on Sd card it always say (file name ).gcode (errno13) permission denied ; (file name)

**Actual results**
can't print with sd card only on USB

**Additional information**
Worked fine last week

[cura.log](https://github.com/Ultimaker/Cura/files/3512582/cura.log)

LipuFei, 2019-08-18:
@fosterswat Could you try to save your gcode to desktop for example and then manually copy the file to your SD card? If you do so, does it succeed, or does it pop up a dialog asking that you need to run as a privileged user/administrator? In case that it asks for administrator's permission, you can follow the steps on this page (https://windowsreport.com/access-is-denied-windows-10/) to fix the permissions for your SD card. It contains very detailed steps.

fosterswat, 2019-08-18:
just found my problem it was really stupid. when i put my sd card in the reader the lock as engaged so i could'nt transfert anything on it thanks for everything next time i will try any simple and stupid thing that could make this if it happen again. i feel so stupid right now to don't have check this.

closed 2019-08-18 by fosterswat

Ghostkeeper, 2019-08-19:
Happens to me every single time. Stupid lock functionality...

syproduction, 2020-12-12:
There may be some plugin failing like Arc Welder. In this case type in Terminal:
`chmod 777 [PLUGIN EXECUTABLE PATH]`

fieldOfView, 2020-12-12:
That's something else entirely. The Arc Welder plugin has not been officially released, and did not even exist in august 2019.