Ultimaker/Cura #6222: Profile does not store all properties

burgi400, 2019-08-18:
**Application version**
4.2.1
**Platform**
Win10

**Printer**
CR-10

**Reproduction steps**
select own profile
change line width to 0.4 (standard 0.44)
update profile with new settings
exporrt profile
close cura
open cura
select own profile

**Actual results**
line width 0.44
**Expected results**
line width 0.4

**Additional information**

LipuFei, 2019-08-19:
@burgi400 I cannot reproduce this issue. After I import the profile and activate it, my line width becomes 0.4 which is the value I set. Could you upload the profile that fails? Thanks!

Note that after you import a quality profile, that quality profile will not be automatically activated. Another problem I noticed with CR-10 is that in Cura 4.2.1 if I import a quality profile that was created with 4.1, select it, and close Cura. The next time I launch Cura, the activated quality profile will be reset to "Standard", but this does not happen with quality profiles created with 4.2.

burgi400, 2019-08-19:
i tried, but it is not a stable behaviour: 
Today i started cura, it shows "Standard Quality" as actual profile with line width 0.4. (should be 0.44)
For some testing i switched to profile "Super Quality", also shown with line width 0.4 (should be 0.44)
After switching back to "Standard Quality" a line width of 0.44 is shown. Switching back to "Super Quality" and also line width 0.44.

LipuFei, 2019-08-19:
Have you modified the line width before you switch the profile? For example, the screenshot below shows that my Line Width has been manually changed to 0.4.

[image]

I can only reproduce this if:
 - I manually change the Line Width to 0.4
 - In the Preferences dialog, change the behavior for profiles to always transfer settings. (see screenshot below)
Then, if I switch between profiles, the Line Width stays at 0.4. I've attached an animation below

[image]

closed 2020-06-19 by no-response[bot]