Ultimaker/Cura #6223: saving profiles incomplete

burgi400, 2019-08-18:
**Application version**
4.2.1

**Platform**
Win 10

**Printer**
CR-10

**Reproduction steps**
open own profile (from cura 4.1)
change settings 
save profile
close cura
open cura
open profile

**Actual results**
settings only partly restored

**Expected results**
all changes are restored

**Additional information**
checking profile in wordpad shows, that only a few attributes are saved, see attechement: one complete profile from 4.1 4kb and a new from 4.2.1 2kb
[GeeetechA30Fein ESPTasteV2.zip](https://github.com/Ultimaker/Cura/files/3512977/GeeetechA30Fein.ESPTasteV2.zip)
[GeeetechA30Fein gayer-andersen Katze.zip](https://github.com/Ultimaker/Cura/files/3512978/GeeetechA30Fein.gayer-andersen.Katze.zip)

LipuFei, 2019-08-19:
@burgi400 I could not reproduce this issue. To be more clear, the following is what I did so you can compare:

 - Open Cura 4.1.
 - I have a CR-10 printer and a custom profile (let's call it `my_profile`). Select or activate `my_profile`.
 - In the custom mode, I change some settings.
 - Click on the profile drop down menu. In the menu, you select "Update profiles with current settings/overrides".
 - Now you have your profile updated, open the quality profile management dialog, and click "Export" to export `my_profile` to a file `my_profile.curaprofile`.
 - Close Cura 4.1.
 - Open Cura 4.2.
 - Make sure that my current machine is CR-10.
 - Open the profile management dialog, and click "Import" to import `my_profile.curaprofile`.
 - `my_profile` appears in the custom profiles list.
 - I select `my_profile` and go to the custom mode to inspect the settings.
 - The values are what I had expected.

burgi400, 2019-08-19:
Sorry, suboptimal description from me.
I used a profile that was created in cura 4.1, missleading.
hope better description:
open cura 4.2.1
select profile which was created wih cura 4.1 (all older profiles shown in listbox, and, in the end, it is also not working with new custom profile created with cura 4.2.1)
change parameters
save as new profile or update existing
export profile
close cura
open cura
select profile -> changes missing, also import of profile does not help

After a few more tests:
it seems that only changed parameters are saved (some times), parameters that are not changed are not saved at all.
This leads to the following behaviour:
if i start with "standard quality" profile, create a new profile and change e. g. layer hight to 0.24 and save the new profile, only this change seems to be respected. If i thenrestart cura, open profile "Super Quality" and then my new profile the initial layer hight is 0.12 instead of 0.2. 
So for reproduacble results i first have to select the profile that was open before creation of the new one and then open new profile.

Ghostkeeper, 2019-08-21:
I'm still not able to reproduce this error. These were the steps I took:

1. Open Cura.
2. Import your profile ESPTasteV2.
3. Activate the newly imported profile.
4. Change Horizontal Offset to 0.1mm. (Horizontal offset is not changed by your profile.)
5. Update existing profile.
6. Export profile. This exported profile is 3.7kB (approximately the same as the original).
7. Close Cura.
8. Restart Cura.
9. Check horizontal offset: Still 0.1mm.

We may be running into [this bug](https://github.com/Ultimaker/Cura/issues/6167) again here though. That always throws mud in the mix.

closed 2020-06-19 by no-response[bot]