Ultimaker/Cura #6224: Cura extremely zoomed in- can't fix it

Jonesy21109, 2019-08-18:
So I tried printing something earlier and as I opened Cura it was so zoomed in, I couldn't see anything. No matter what I did (deleted, reopened, tested all settings) it would not go back to how it should be. Does anyone have any advice? Thank you so much in advance. 
[image]

LipuFei, 2019-08-19:
Hi @Jonesy21109 , does this happen for you only with Cura 4.2.1, or also previous versions such as 4.1 and 4.0? we have also seen some similar issues before that are related to graphics cards and drivers. Could you let us know which CPU, GPU, and GPU driver version(s) you are using? Thanks.

Ghostkeeper, 2019-08-19:
Could you try changing the DPI settings for Cura? Cura is rescaling the UI based on what it thinks your system's DPI is. Here is a tutorial: https://www.tenforums.com/tutorials/15523-change-compatibility-mode-settings-apps-windows-10-a.html#option1

closed 2020-06-19 by no-response[bot]