Ultimaker/Cura #6225: where is gpg signing key C1A1B91069C4AF59 ?

sloth1, 2019-08-19:
Looking for gpg signing key C1A1B91069C4AF59 used to create:
 Ultimaker_Cura-4.2.1.AppImage.asc

LipuFei, 2019-08-19:
The public key is now on the public key servers.
```
$ gpg --keyserver pgp.mit.edu --recv-keys C1A1B91069C4AF59
gpg: /home/l.fei/.gnupg/trustdb.gpg: trustdb created
gpg: key C1A1B91069C4AF59: public key "Ultimaker Build Server <monolith@ultimaker.com>" imported
gpg: Total number processed: 1
gpg:               imported: 1
```

closed 2019-08-19 by LipuFei

sbrl, 2021-09-12:
@LipuFei: That returns an error for me:

```
$ gpg --keyserver pgp.mit.edu --recv-keys C1A1B91069C4AF59
gpg: keyserver receive failed: Server indicated a failure
```

ElectricRCAircraftGuy, 2022-07-31:
I'm getting an error: 

```bash
$ gpg --keyserver pgp.mit.edu --recv-keys 1889EAED2DB93BF7DFFB3CA6C1A1B91069C4AF59
gpg: keyserver receive failed: Server indicated a failure
```