Ultimaker/Cura #6226: Spiralize produced incorrect results in 4.2.1 (worked in an older version)

opcow, 2019-08-19:
**Application version**
4.2.1
**Platform**
Windows 10 1903

**Printer**
Ender 3 Pro

**Reproduction steps**

1. Load pyramid4.stl
2. Click the slice button
3. Observe the result in preview

**Actual results**
Most of the "layers" are skipped, resulting in incomplete and unprintable gcode.

**Expected results**
Older versions of Cura were able to produce good gcode for printing this object. Other slicers I have tried also are able to handle spiral/vase/corkscrew mode using this object.
 
**Additional information**
[image]

[CE3_pyramid4.zip](https://github.com/Ultimaker/Cura/files/3514171/CE3_pyramid4.zip)

smartavionics, 2019-08-19:
You must either reduce the line width to 0.35 or less or scale the model up for it to slice OK.

[image]

smartavionics, 2019-08-19:
I printed one at 200% scale and it came out fine.

opcow, 2019-08-19:
Interesting. I saw case where selecting a 1mm nozzle size made the walls of a box disappear completely when sliced unless the line width was reduced. I assume that was related to this issue.

Ghostkeeper, 2019-08-19:
I wonder what part of this model makes it fail with spiralise though at the 0.44mm line width. When spiralise is disabled, the outer walls come out fine.

smartavionics, 2019-08-19:
It's because when spiralizing only 1 part from each layer is printed but when not-spiralizing all parts on each layer are printed.

smartavionics, 2019-08-19:
It's quite amazing that this model can be printed using a single line and no travels. The marvel of 3d printing!

smartavionics, 2019-10-02:
> IMHO, Cura has a problem with thin walls and 1-wall models, spiralized or not. This is really weird, and makes it unusable for thin models printed with large nozzles, when nozzle = wall width. But it prints perfectly if the nozzle is half the wall thickness (takes too long !)

Yes, that's right. It's a fundamental problem with how Cura processes walls. I don't know if the developers have any plans to reimplement how walls are processed.

yet-another-average-joe, 2019-10-02:
@smartavionics  Is it so deep (low level) into the code that you can't solve the problem in a future version of your fork without killing everything else ? Maybe a special mode ?

smartavionics, 2019-10-02:
> @smartavionics Is it so deep (low level) into the code that you can't solve the problem in a future version of your fork without killing everything else ? Maybe a special mode ?

Yes, it's pretty fundamental. I would hope that the Cura developers have a plan to redo how the walls are processed but I suspect that is not a very high priority goal for them given how long it has been how it is.

yet-another-average-joe, 2019-10-03:
I experimented further. First, I printed with slic3r, and got terrible results with 0.8 nozzle + 0.8 walls. But I i'm not used to this slicer. The first I tried, but I immediately switched to Cura (today was my very first print with slic3r, and probably the last one !)

Then I played again with walls. I didn't have time to test with @opcow models. Tomorrow... Setting a line width slightly larger than the nozzle drastically changes Cura's behaviour. This overextrusion can be compensated with flow rate or filament diameter. Just have a look to these images :

wall = nozzle (different patterns and gaps appear depending on other parameters) :

[image]

Now, line width slightly > nozzle : LOOKS GOOD !!! :

[image]

To be tested on OP files...

Could all these glitches depend on some rounding errors ? Does the slicer core use some fixed point arithmetics ? I recall similar glitches (2D, not 3D) on FP calculations for scrolling/zooming buffered viewports with Win32 API. Tiny error + tiny error -> a hole (a missing 1 pixel line or one more than needed).

smartavionics, 2019-10-03:
The yellow walls are walls that have been printed using the thin walls feature. Thin walls (when enabled) are printed where there is a wall whose width is <  wall line width.

Ghostkeeper, 2019-10-03:
The nozzle size is hardly used by Cura at all. All that matters is the line width vs. how wide the walls are. If the line width is exactly the same as the model's width, then due to rounding in the STL file and rounding by Cura it'll most likely be sometimes slightly more than the model's width and sometimes less. This causes the issue that yet-another-average-joe started with. Other slicers have threshold problems like this as well but the thresholds lie in different widths so it's not an argument that it doesn't happen with your model with their default settings.

We do have some research going on with varying line widths to make thin walls fit better without requiring the hatching technique. At this point that is only in the research stage though. It has some fundamental problems with what sort of things a Bowden style printer is able to extrude.

yet-another-average-joe, 2019-10-03:
Thanks for this answer. I was suspecting rounding issues (I edited my message, removing the question). I now better understand : the slicer just deposits some material, assuming it will spread evenly in the space between the nozzle and the previous and adjacent layers.

I wanted to see the OP's problem, I installed 4.2.1, but it crashes before the main window shows up (4.1 and 4.3 were previously installed...). Therefore, I can't test the trick I use and that gives the expected results for my problem. Could someone give it a try ? Slightly larger line width + thin walls activated ? (no problem with 4.1 and 4.3)

About the model I was showing : Cura gave a perfect print with this trick (except some holes because I forgot to make the seam random, and had a 6.8mm retraction on a full metal E3D Volcano - didn't verify these settings 8) ). The other slicer did show a perfect preview, but produced an horrible print (underextrusion here and there with very little retraction). I'm interested in all this little issues because I have 96 such models to make (still underoptimal separators for drawers - electronics parts management)

I imagine the complexity of slicer engine development : intersecting triangles with a plane, getting contours, calculating surface fillings, and optimizing toolpaths (I was a dunce at topology courses, BW killed me :) ). Members and contributors are math and code gods !

GregValiant, 2023-09-15:
Since the Arachne engine has now been rolled in can this be closed?

closed 2023-09-29 by github-actions[bot]