Ultimaker/Cura #6228: Heating bed no longer works after using ChangeAtZ on Cura 4.2.1

GodChronoss, 2019-08-19:
**Application version**
4.2.1

**Platform**
Windows 10, AMD Ryzen Threadripper 2950X, 128 GB ram, Nvidia Geforce RTX 2080 TI 11GB OC

**Printer**
Creality CR10S pro with TynyMachines 3D firmware 1.1.9_B7 

File Used: [CCR10S_print.zip](https://github.com/Ultimaker/Cura/files/3516694/CCR10S_print.zip)

**Reproduction steps**
I imported a model to be 3D printed, went to Extensions --> Post Processing --> Modify G-Code and added a ChangeAtZ 5.1.1 (experimental) script. I used the following:

Trigger: Layer No.
Change Layer: 5
Behavior: Keep Value
Layer Spread: 1
Change Bed Temp: Checked
Bed Tem: 0

Then just slice and print.

**Actual results**
The sliced file worked fine on the first printer, but I had to stop it and restart it 2 times (my fault for not calibrating the bed the first time and had to use some glue stick the second time because the print was curling up). on the third try, the printer gave me a 'thermal runaround' error message and now the heating bed does not want to heat up.

The second printer was using the same file (bed was calibrated, didn't use glue stick). I stopped it after 9 hours because the print was curling up (never had this problem before while printing on mirror tiles and a heating bed). Now this printer is suffering from the same problem when I try and restart the print. The bed is not heating up at all!

 I have tried several times to heat up the bed of the 2 printers with no luck. I have waited 10, 15, 20 mins with the printers unplugged. I plugged them back in, set up the bed to heat up and nothing happens each time. 

**Expected results**
well.. the bed should heat up after restarting the printer

**Additional information**
I don't think 2 printers would have failed at the same time from using the same file. I'll keep troubleshooting the printers but I keep thinking the experimental code might have something to do with this problem. Oh and the printers are only a few months old and barely use them.

GodChronoss, 2019-08-20:
I fixed it. Thanks

closed 2019-08-20 by GodChronoss

Ghostkeeper, 2019-08-21:
Maybe it'd be useful for people to know how you fixed it, in case they also encounter this problem.

GodChronoss, 2019-08-21:
Oh sorry! I didn't know why the whole thing was deleted.

The problem with my 2 printers happened at the same time I was testing the ChangeAtZ script for the fist time. The gcode turned off my bed at the layer I wanted. The problem was the heating beds never tuned back on after that.

I was blaming the gcode for the failures. I was wrong!. It turned out it was the wiring on both my CR10s PRO's heating bed that got damaged while printing (one melted and the other one almost catches fire during troubleshooting). I ended up rewiring the beds with a 16 gauge teflon coated wire to fix both printers. 

I had it all explained in the post that I guess I deleted by accident. I don't know how it happened. I was really tired when I posted the fix. Sorry!