Ultimaker/Cura #6229: [4.2.1] Enabling support disables build plate adhesion

theWaldschrat, 2019-08-19:
**Application version**
4.2.1

**Platform**
Windows 10

**Printer**
Creality Ender 3, i.e. from the new Creasome Mod printers

**Reproduction steps**
Open Cura 4.2.1, select above printer, select any profile, toggle "Generate Support" on and off and on and off...

**Actual results**
With support on, build plate adhesion is set to none. Switching support back sets it back to skirt.

**Expected results**
Changing support settings doesn't change build plate adhesion settings (as in versions prior to 4.2).

**Additional information**
I don't think anyone expect support setting to have an impact on build plate adhesion. At least I didn't. And so I got confused.
Also I don't think it's any helpful, because with skirt off (and the user not noticing if she doesn't check), the print starts with the support and if the nozzle isn't properly primed, the support won't be attached properly to the bed. And supports aren't exactly known to stick well to the build plate anyway.

rburema, 2019-08-20:
It seems that adhesion is off by default, if I toggle it on, I can't reproduce the problem as I understand it:
[image]
The adhesion checkmark stays in the checked position (and does generate brim adhesion if checked).

Maybe there's something I don't understand about your report though.

theWaldschrat, 2019-08-20:
OK, using the "Recommended" print setting view, things are even more confusing. It does the same as in the "Custom" view, but you don't even see it.
Anyway, back to the actual problem. I removed my Cura profiles and started fresh to be sure things weren't caused by my profiles.
Steps to reproduce once more:
- Start Cura (fresh profile).
- Select Creality Ender 3
- Load a model.
- Slice.
- Lo and behold, a skirt is printed.
- Check "Support"
- Slice.
- Lo and behold, no skirt is printed.
- Uncheck "Support".
- Change view to "Custom".
- Look in "Support" section: Nothing ticked.
- Look in "Build Plate Adhesion" section: "Skirt" is selected.
- Tick "Generate Support".
- Observe "Skirt" change to "None".
- Untick "Generate Support".
- Observe "None" change back to "Skirt".

Did I miss anything?

Ghostkeeper, 2019-08-21:
No, those reproduce steps are very clear. I can reproduce it too.

However this is expected behaviour. It is caused by this profile:

https://github.com/Ultimaker/Cura/blob/9122d0cca44d684d937898afa6aa7b39e5180d02/resources/definitions/creality_base.def.json#L110

The adhesion type is explicitly set to None if the support is enabled. This is on purpose. Apparently the skirt is unnecessary if you have support. Maybe support is not as sensitive to bed levelling.

closed 2019-08-21 by Ghostkeeper

Liger0, 2019-08-21:
Please remove it, it's extremely stupid. Especially when supports are used skirt is needed.

Ghostkeeper, 2019-08-22:
Yeah, I'll concur. I believe that enough people have tested this. And it's confusing too. I've made it always use skirt now.

theWaldschrat, 2019-08-22:
Nice, thanks, spares me from writing lengthy comments about skirts having more to do with nozzle priming than bed leveling.

mu2004, 2019-09-09:
When the support is NOT built from the bed, but rather used on a higher layer, it makes absolutely no sense to disable skirt!

commit 9f169732 (2019-08-22): Don't remove skirt when printing with support

According to tests by theWaldschrat and Liger0, the skirt is still necessary when support is enabled.

Fixes #6229.