Ultimaker/Cura #6230: 4.2.1 single pass round features split into multiple passes

Wdfitzpat007, 2019-08-19:


smartavionics, 2019-08-19:
Hi, I don't really understand what the problem is from what you have written above but if you could save the project file (File -> Save) and then either zip or rename the .3mf to .zip and attach to this issue then I can take a look and maybe help. Thanks.

Ghostkeeper, 2019-08-21:
I'd not expect a contour to be broken off halfway and continued later. Is that what you meant?

As for the order in which the rings are printed, it could well look very arbitrary because it's trying to minimise travel moves with Nearest Neighbour. It's not in the same organised way that a human would order these parts because a computer is not able to understand that concentric rings belong to each other. It just looks at travel distances.

closed 2020-06-19 by no-response[bot]