Ultimaker/Cura #6231: 4.2.1All print settings can not be modified

Acccl, 2019-08-20:
**Application version**
4.2.1

**Platform**
win10x64

**Reproduction steps**
Sometime I want to modify  values of print settings,but I find them can not be edited .when I try to edit a value ,the value can be edited,But after editing,the value always revert.
**Actual results**
the value always revert.
**Expected results**
Pirnt settings can be modified  at any time.

**Additional information**

closed 2019-08-21 by Acccl

Ghostkeeper, 2019-08-21:
I'm not sure if you intended to close your own ticket. But I'm not able to reproduce this bug in any case.

Acccl, 2019-08-22:
I have found the  solution which is related to my setting of the input method. I am a user from china,I find if I use the English mode setting instead of Chinese mode setting,all value of print setting will be edited.Otherwise，I can't modify my print setting.

Ghostkeeper, 2019-08-22:
Hmm, so maybe your keyboard input is trying to put the fullwidth characters in there instead of the ordinary ASCII numerals?

Acccl, 2019-08-22:
yes,I think so.