Ultimaker/Cura #6232: [4.2.1 & 4.1.0]Cura Slicing Adds Block Of Material To Model When Rotated

xantos117, 2019-08-20:
**Application version**
[4.2.1] and [4.1.0] both have the error occur

**Platform**
Windows 10 x64, issue has occurred on 3 machines, each with different hardware. 

**Printer**
Ultimaker-007eec

**Reproduction steps**
1. Import skull stl into Cura
2. Rotate skull model to have flat top as base of print
3. Slice and preview model

**Actual results**
Slicing generates a large filled block of material within the skull that isn't part of the model. Additionally, a number of layers are missing. This second part is rotation invariant, and the missing layers are always parallel to the printing surface. 
These errors occur whether support material is selected or not, irrespective of the material.

**Expected results**
The sliced skull is the same as the stl model, without added material. All slices are present.

**Additional information**
Zip of stl file to be printed: [006 trimmed mandible removed 2 - skull base - support arch top removed - sealed.zip](https://github.com/Ultimaker/Cura/files/3521558/006.trimmed.mandible.removed.2.-.skull.base.-.support.arch.top.removed.-.sealed.zip)
Zip of Cura Project: 
[UMS5_006 trimmed mandible removed 2 - skull base - support arch top removed - sealed.zip](https://github.com/Ultimaker/Cura/files/3521563/UMS5_006.trimmed.mandible.removed.2.-.skull.base.-.support.arch.top.removed.-.sealed.zip)
Missing print layers: 
[image]
Skull positioning: 
[image]
Skull after slicing, notice the filled section in the center: 
[image]

Filled block doesn't appear when skull rotated to other orientations, but missing layers appear no matter what. 

This skull stl is a modified version of a larger skull that I don't believe I can easily paste in to this report. I modified it by passing rectangular prisms through the original and using a boolean subtraction to remove material. Once parts were separated, I could remove them from the model, and they were thus deleted. Inspection of verticies and faces in Blender show no oddities that I can figure out would cause this issue.

smartavionics, 2019-08-20:
The model has problems. It is unlikely to be sliced well until those problems are fixed. Sorry, I don't know how you would do that.

[image]

Ghostkeeper, 2019-08-21:
The model is open on the inside. You can see the open insides where there is just a sheer cut out from the model. It is ambiguous then what parts of the model are inside the model and what parts are outside. Cura cannot fix this for you, because it doesn't know what you want printed, though it tries to mend it as good as it can.

There is nothing we can realistically do to prevent this. Mesh fixing is outside of Cura's scope. Sorry!

closed 2019-08-21 by Ghostkeeper