Ultimaker/Cura #6233: cura 4.2.1 will not slice

DrDzSlaughter, 2019-08-21:
ever since I added in the new update to 4.2.1 everything works like it should all the way until I get to the slicing part then the bar fills a 1/10th  of the way then just stops I went through firewall seeing if maybe it was blocking it thinking it was a malware, and that still didn't fix it I went back to the 4.1.0 program I had before the update and it also is doing the exact same thing... I really love using cure for my 3d printing but haven't been able to splice anything in close to a month, I was hoping with time a patch would have maybe fixed it but haven't read anything on this problem please help

LipuFei, 2019-08-21:
Hi @DrDzSlaughter Could you upload a project file for your model and settings? You can do that by using the menu `File` -> `Save`.

Liger0, 2019-08-21:
Yeah in 4.2 and 4.2.1 slicing and preview are extremely slow as no hardware resource is used.

roydaman2, 2019-10-15:
I'm having the same issue... doesn't matter if I wait 5 mins or 5 hours slicing never completes... I will get "project file" as soon as I get home from work...

closed 2020-06-19 by no-response[bot]