Ultimaker/Cura #6234: cura 4.2.1 : print hangs on large surface print

JLMorange, 2019-08-21:
**Application version**
4.0 to 4.2.1

**Platform**
windows7 et windows10

**Printer**
ultimaker original
see attached project

**Reproduction steps**
print the attached generated gcode, it will hang at first layer

**Actual results**
the printer stop, print head's temperature remain at requested level, the printer does not respond (tried to home or move head position)

**Expected results**
the print should have finished

**Additional information**

JLMorange, 2019-08-21:
generated gcode and project
[semelleMakita.zip](https://github.com/Ultimaker/Cura/files/3524940/semelleMakita.zip)

JLMorange, 2019-08-21:
printer state when hang occurs
[image]
[image]

Ghostkeeper, 2019-08-21:
Maybe there is a bad spot with a lot of friction on your axles there? I don't see anything in the g-code that could cause this.

JLMorange, 2019-08-22:
doing the same slice in another slicer do perform the print, printing another small surface file at this location did success, so it's not a printer issue.... it's really related to cura... maybe a overflow (cura seems to not stop the print when I ask it while it do the surface zigzag, but pressing home it immediately performed. So it looks like the printer has some commands, will take the stop when finishing the command, but perform the go home right away)? is there a debug communication trace window in cura?

Ghostkeeper, 2019-08-22:
Are you printing via USB cable or do you have the SD card upgrade?

There used to be a bug where Cura could keep sending commands through USB while the printer was heating up, but I believe that was solved a few releases ago.

JLMorange, 2019-08-22:
printing through usb

colatuttorf, 2019-08-22:
I had this same issue with parts posted Cura 4.1.  Updating to 4.2.1 and changing no additional settings seems to have solved the issue, so far at least.

JLMorange, 2019-08-28:
still wonder if there is a way to have a trace of communication between cura and printer?

Ghostkeeper, 2019-09-03:
You could trace all communication with a COM port sniffer like this one: https://www.eltima.com/products/serial-port-monitor/

JLMorange, 2019-09-23:
the tool you gave me isn't able to monitor while cura locks the com port... I don't found any tool that can open the com port while cura is printing.... don't you think it could be easier to create either a debug window or duplicate to a file???

Ghostkeeper, 2019-09-24:
You can duplicate to a file by selecting "Save to File" before you print, though that doesn't really allow you to read where it stops.

If you don't trust that the USB file writer is writing the same g-code, you could duplicate the g-code to the log from the writer by modding your Cura installation on Windows. Here's how:
* Copy the file `C:\Program Files\Ultimaker Cura 4.3\plugins\USBPrinting\USBPrinterOutputDevice.py` to your desktop.
* Open it with a text editor.
* At line 383, just before `self._sendCommand`, insert this line: `        Logger.log("d", line)` (note that there are 8 spaces before it).
* Save the file and copy it on top of the `USBPrinterOutputDevice.py` file in your Program Files where you got it from. You'll need admin access on your computer for this step.

This will put the entire g-code in your Cura log file. It's not advisable to try this for very long or with large g-codes.

closed 2020-07-06 by no-response[bot]