Ultimaker/Cura #6236: [4.2.1] Materials from marketplace do not show in the materials lists

bytespider, 2019-08-21:
[cura.log](https://github.com/Ultimaker/Cura/files/3526193/cura.log)

**Application version**
4.2.1

**Platform**
MacOS

**Printer**
Ender-3 [Ender-3.3mf.zip](https://github.com/Ultimaker/Cura/files/3526200/Ender-3.3mf.zip)

**Reproduction steps**
Go to marketplace
Add materials (Filamentive)
Click "Quit Cura" button presented in the dialog
Re-Open cura
Click "Manage materials"

**Actual results**
See no new materials
From the filament dropdown, see no new materials

**Expected results**
See Filamentive materials that were installed from Marketplace

**Additional information**
[image]
[image]

LipuFei, 2019-08-21:
Many material profiles in the Marketplace are with 2.85mm diameter. In the screenshot, I see that your machine uses 1.75mm materials. You can add another printer, for example, Ultimaker S5, which uses 2.85mm materials. Then check your material list see if the materials you installed from the Marketplace appear. Could you check that?

bytespider, 2019-08-21:
You are correct, the do show up when a nozzle of 2.85 is selected.

I guess the issue is making that clear to the user what diameter the materials are compatible with.

mentor2k1, 2021-07-13:
For everyone who landed here: 
Little trick to use marketplace materials with 2.85 also in printers with 1.75:

1. Change the material diameter in your printer to 2.85
2. Go to the marketplace material
3. export the fdm_material file
4. change in the xml the diameter to 1.75 and the name of the material
5. Change the material diameter in your printer to 1.75
6. Import the fdm file