Ultimaker/Cura #6237: Setting corner speed (slowdown)

kyjak, 2019-08-21:
Using larger nozzles with higher printing speeds destroys the quality
of sharp corners (90 degree is a common one for example). The filament has no time
to stick in the sharp corner very well.

I would suggest a parameter to slow down print on angles sharper than a value.
So 2 values - angle threshold and a percentage of downspeed

The only fix now is to set a small speed to outer wall, however, having few corners
makes printing any straights equally slow, thus delaying the print unnecessary.

All printers are affected (I own a prusa mk3)
Everybody would benefit, any advanced user.

smartavionics, 2019-08-21:
Have you tried using lower acceleration values for the walls?

Ghostkeeper, 2019-08-22:
There is a setting for this: Jerk. Jerk is the speed that the nozzle is allowed to run through corners. Sharp corners get slowed down more than obtuse corners. I advise that you lower the jerk setting considerably. For example, Ultimaker 3 uses an outer wall jerk of 5mm/s.

For the Prusa i3 you first need to check the Enable Jerk Control setting so that Cura takes over control of how much jerk to use during your print. Otherwise it lets the firmware decide.

closed 2019-08-22 by Ghostkeeper

kyjak, 2019-08-22:
Thank you, acceleration did the trick. Sorry for the ticket.