Ultimaker/Cura #6239: [4.2.1] Segmentation fault (SEGFAULT) after launch

chhu, 2019-08-22:
**Application version**
4.2.1 AppImage

**Platform**
RHEL 7.6 x64
NVIDIA 418.39

**Printer**
UM3E (not connected)

**Actual results**
After download, I started, I see the GUI for a few seconds but cannot click anything, then SEGFAULT crash. Deleted previous profiles, I see the welcome window, again crash. :(

**Additional information**
```
2019-08-22 14:00:47,367 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin ConsoleLogger
2019-08-22 14:00:47,388 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin CuraDrive
2019-08-22 14:00:47,418 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend.__init__ [78]: Found CuraEngine at: /tmp/.mount_curajubA2m/usr/bin/CuraEngine
2019-08-22 14:00:47,428 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin CuraEngineBackend
2019-08-22 14:00:47,443 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin CuraProfileReader
2019-08-22 14:00:47,455 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin CuraProfileWriter
2019-08-22 14:00:47,467 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin OBJReader
2019-08-22 14:00:47,478 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin OBJWriter
2019-08-22 14:00:47,483 - WARNING - [MainThread] STLReader.STLReader.<module> [26]: Could not find numpy-stl, falling back to slower code.
2019-08-22 14:00:47,492 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin STLReader
2019-08-22 14:00:47,503 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin STLWriter
2019-08-22 14:00:47,515 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin FileLogger
2019-08-22 14:00:47,530 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin FirmwareUpdateChecker
2019-08-22 14:00:47,544 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin FirmwareUpdater
2019-08-22 14:00:47,556 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin GCodeGzReader
2019-08-22 14:00:47,567 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin GCodeGzWriter
2019-08-22 14:00:47,579 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin GCodeProfileReader
2019-08-22 14:00:47,602 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin GCodeReader
2019-08-22 14:00:47,614 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin GCodeWriter
2019-08-22 14:00:47,631 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin ImageReader
2019-08-22 14:00:47,645 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin LegacyProfileReader
2019-08-22 14:00:47,659 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin LocalContainerProvider
2019-08-22 14:00:47,673 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin LocalFileOutputDevice
2019-08-22 14:00:47,687 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin MachineSettingsAction
2019-08-22 14:00:47,699 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin ModelChecker
2019-08-22 14:00:47,751 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin MonitorStage
2019-08-22 14:00:47,777 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin PerObjectSettingsTool
2019-08-22 14:00:47,792 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin PostProcessingPlugin
2019-08-22 14:00:47,804 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin PrepareStage
2019-08-22 14:00:47,816 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin PreviewStage
2019-08-22 14:00:47,830 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin RemovableDriveOutputDevice
2019-08-22 14:00:47,862 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin SimulationView
2019-08-22 14:00:47,878 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin SliceInfoPlugin
2019-08-22 14:00:47,891 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin SolidView
2019-08-22 14:00:47,906 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin SupportEraser
2019-08-22 14:00:47,926 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService.getAccessToken [112]: No auth data to retrieve the access_token from
2019-08-22 14:00:47,935 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin Toolbox
2019-08-22 14:00:47,949 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin CameraTool
2019-08-22 14:00:47,965 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin MirrorTool
2019-08-22 14:00:47,981 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin RotateTool
2019-08-22 14:00:47,998 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin ScaleTool
2019-08-22 14:00:48,010 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin SelectionTool
2019-08-22 14:00:48,026 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin TranslateTool
2019-08-22 14:00:48,046 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin UFPReader
2019-08-22 14:00:48,064 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin UFPWriter
2019-08-22 14:00:48,168 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin UM3NetworkPrinting
2019-08-22 14:00:48,203 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin USBPrinting
2019-08-22 14:00:48,218 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin UltimakerMachineActions
2019-08-22 14:00:48,228 - INFO - [Thread-1] UpdateChecker.UpdateCheckerJob.run [36]: Checking for new version of cura
2019-08-22 14:00:48,242 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin UpdateChecker
2019-08-22 14:00:48,278 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin VersionUpgrade21to22
2019-08-22 14:00:48,299 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin VersionUpgrade22to24
2019-08-22 14:00:48,320 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin VersionUpgrade25to26
2019-08-22 14:00:48,339 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin VersionUpgrade26to27
2019-08-22 14:00:48,356 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin VersionUpgrade27to30
[... 217 lines cut ...]
  File "/build/4.2/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/build/4.2/build/inst/lib/python3.5/site-packages/UM/JobQueue.py", line 112 in run
  File "/build/env/4.2/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/4.2/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007fdf5979f740 (most recent call first):
  File "/build/4.2/build/inst/lib/python3.5/site-packages/UM/Qt/QtApplication.py", line 341 in exec_
  File "/build/4.2/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 799 in run
  File "/build/4.2/build/inst/bin/cura", line 137 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "/build/env/4.2/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/Console.py", line 21 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "/build/env/4.2/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load

```

chhu, 2019-08-22:
Closing firefox while starting helped (?!)

chhu, 2019-09-10:
Sorry, it is back and worse, tried switching QT versions, everything. Now crashes every time. No clue, all other programs are working normally...
4.2.0 beta has the same problem, 4.1.0 runs fine.

I hope I'm not the only one with this problem...

Ghostkeeper, 2019-09-10:
A Qt version is included in the AppImage, so switching Qt versions on your system should not have any influence.

From the logs, it can be seen that the main thread is executing QML code at the moment, so it looks like the segfault is happening in Qt itself.

You could try running the AppImage through GDB, maybe?

chhu, 2019-09-11:
Decompressed and ran through gdb, seems libcrypto is the bad guy...
```
Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fff7f2f4700 (LWP 3046)]
0x00007ffff7a9e365 in sk_num () from /usr/lib64/libcrypto.so.10
(gdb) where
#0  0x00007ffff7a9e365 in sk_num () from /usr/lib64/libcrypto.so.10
#1  0x00007ffff7ae2f6c in obj_trust () from /usr/lib64/libcrypto.so.10
#2  0x00007ffff7add09c in X509_verify_cert () from /usr/lib64/libcrypto.so.10
#3  0x00007fffee983898 in ssl_verify_cert_chain () from /scr2/cdec/usr/bin/libssl.so.1.0.0
#4  0x00007fffee95b5ec in ssl3_get_server_certificate () from /scr2/cdec/usr/bin/libssl.so.1.0.0
#5  0x00007fffee96062a in ssl3_connect () from /scr2/cdec/usr/bin/libssl.so.1.0.0
#6  0x00007fffee96b006 in ssl23_connect () from /scr2/cdec/usr/bin/libssl.so.1.0.0
#7  0x00007fffe24a78e5 in ?? () from /scr2/cdec/usr/bin/libQt5Network.so.5
#8  0x00007fffe24a84de in ?? () from /scr2/cdec/usr/bin/libQt5Network.so.5
#9  0x00007fffe248c601 in ?? () from /scr2/cdec/usr/bin/libQt5Network.so.5
#10 0x00007fffe51df3a7 in QMetaObject::activate(QObject*, int, int, void**) () from /scr2/cdec/usr/bin/libQt5Core.so.5
#11 0x00007fffe245c3f3 in ?? () from /scr2/cdec/usr/bin/libQt5Network.so.5
#12 0x00007fffe245c4ac in ?? () from /scr2/cdec/usr/bin/libQt5Network.so.5
#13 0x00007fffe246bfc1 in ?? () from /scr2/cdec/usr/bin/libQt5Network.so.5
#14 0x00007fffe2d9bc8c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /scr2/cdec/usr/bin/libQt5Widgets.so.5
#15 0x00007fffe2da2f01 in QApplication::notify(QObject*, QEvent*) () from /scr2/cdec/usr/bin/libQt5Widgets.so.5
#16 0x00007fffe344ed0e in sipQApplication::notify(QObject*, QEvent*) () from /scr2/cdec/usr/bin/lib/python3.5/PyQt5/QtWidgets.so
#17 0x00007fffe51b3328 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /scr2/cdec/usr/bin/libQt5Core.so.5
#18 0x00007fffe520916d in ?? () from /scr2/cdec/usr/bin/libQt5Core.so.5
#19 0x00007fffe4a5b049 in g_main_context_dispatch () from /lib64/libglib-2.0.so.0
#20 0x00007fffe4a5b3a8 in g_main_context_iterate.isra.19 () from /lib64/libglib-2.0.so.0
#21 0x00007fffe4a5b45c in g_main_context_iteration () from /lib64/libglib-2.0.so.0
#22 0x00007fffe520850b in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /scr2/cdec/usr/bin/libQt5Core.so.5
#23 0x00007fffe51b1b8a in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /scr2/cdec/usr/bin/libQt5Core.so.5
#24 0x00007fffe4fdb64b in QThread::exec() () from /scr2/cdec/usr/bin/libQt5Core.so.5
#25 0x00007fffe4fe066c in ?? () from /scr2/cdec/usr/bin/libQt5Core.so.5
#26 0x00007ffff7764dd5 in start_thread () from /lib64/libpthread.so.0
#27 0x00007ffff6b7d02d in clone () from /lib64/libc.so.6
(gdb) 

```

chhu, 2019-09-11:
Fixed. I had an issue before with a broken libcrypto that was included in a similar package (not cura), so I used LD_PRELOAD=/usr/lib64/libcrypto.so to force using native libcrypto. Removing that and cura works flawless. Thanks for the hint!

closed 2019-09-11 by chhu

Ghostkeeper, 2019-09-11:
Maybe we need to include libcrypto with the AppImage then? Do you think that is a good idea, @LipuFei?