Ultimaker/Cura #6241: Crash after sleep on notebook

userosos, 2019-08-22:
Hi! I can see crash the cura 4.2.1 if i use the notebook with cnanger after it i close the notebook (it goto sleep) after it i open it without changer (netbook goto green mode in widnows 10). I think that after sleep notebook use intagation video adapter (in green mode all apps switch to integration video adapter). I started cure use external video adapter. My netbook is Lenovo thinkpad p52.

rburema, 2019-08-23:
Yes, Cura probably needs your external video adapter. Does this also happen when you explicitly tell windows to run the app with your external adapter?

Before we go into that direction though, we should make sure that this is really what's going on. Can you send us the logs? They're in `%appdata%/cura/4.2`.

userosos, 2019-08-24:
@rburema 
>Does this also happen when you explicitly tell windows to run the app with your external adapter?
Yes.
Logfile - https://www.sendspace.com/file/sm4dtf
But it is very big file.

Ghostkeeper, 2019-08-25:
That's indeed a big file! Looks like that Repetier plug-in put almost 150MB of logs in about 2 weeks in there.

Incidentally, the last 187 lines of log before the crash are about the Repetier plug-in. It seems to be doing stuff with requests to LAN. Could be that something crashes there after sleep mode (because it sleeps before getting a reply or something segfaults in a camera stream?) Does the problem still occur if you temporarily disable the plug-in?

closed 2020-06-19 by no-response[bot]