Ultimaker/Cura #6242: Spiral model placment

LD3D, 2019-08-22:
**Application version**
<Cura 4.2.1>

**Platform**
<windows 10>

**Printer**
<CR10S-MAX>

**Reproduction steps**
<In Preview mode after slicing>

**Actual results**
<Models will print at random locations on the build bed>

**Expected results**
<Set the order of the printed models>

**Additional information**
<A simple left-click or right-click and select the print order according to the amounts of models on the build bed>
[Spiral mode.curaproject.3mf.zip](https://github.com/Ultimaker/Cura/files/3532294/Spiral.mode.curaproject.3mf.zip)

smartavionics, 2019-08-23:
You don't provide any description of the problem and I cannot load your project file because the printer type is unknown so I haven't a clue what your issue is. I do see multiple models so one comment I have is that you cannot spiralize more than one model at a time.

Ghostkeeper, 2019-08-23:
There was information, but the `

smartavionics, 2019-08-23:
Ah, OK. And I think now that the issue is not to do with spiralization mode but the placement of the models.

Ghostkeeper, 2019-08-23:
I also can't load the project file. You've modded your Cura version with an additional printer and that's the one being used in this project file. Can you provide the files necessary to mod Cura in the same way with the CR10s Max printer?

I do see that he's using One At A Time mode though, so spiralize shouldn't be a problem.

The order of the models in one at a time mode cannot be configured because it's decided by Cura in order to minimise the collision area of each object to the theoretical optimum, as long as Cura knows the correct print head shape. If Cura would allow the user to change the order, then the collision area for each object is different for each other object it can collide with, basically preventing you from knowing what the collision area really is.

The current order is determined in order to minimise the collision areas. However if models are sufficiently far apart that either direction works, then the order is arbitrary based on how the algorithm works.

closed 2019-08-23 by Ghostkeeper

LD3D, 2019-08-23:
Team, thank you for the fast response. The problem is that the spiral mode wants to always print in a random order if more then 1 model is on the build plate regardless of the printer being used. 

The User should be able to pick the order the print happens in one at a time printing. in the first place to have Cura do this at random is not a great solution and it slows down the printing and if models are of different heights or sizes Cura will damage them because it wants to print at random.

2 possible solutions are the only fixes for this:
#1 Let the user specify what model prints first and the order of prints by having a selection checkbox for the order to print in a dropdown menu on a mouse click right or left it does not matter.
or
#2 the Print will happen from the home position or the first model inserted into the build area and will print each model after that as added to the build plate. So if you add 1 model it will print that one but then you add 2 so it will print 1 and then 2 If you add a 3 then it will print 1 and then 2 and then 3 and so on.

this is a really easy request to fix a collision due to a print order issue.

Ghostkeeper, 2019-08-25:
Cura does NOT print objects in a random order! The order is determined using this algorithm here: https://github.com/Ultimaker/Cura/blob/master/cura/OneAtATimeIterator.py
This directly clashes with your proposed changes, so we can't implement them.

If Cura is causing your print head to collide with previously printed objects in one-at-a-time mode, then the [shape for the print head](https://github.com/Ultimaker/Cura/blob/8d8f18d9535201543b6031903f73cc4e7b2bbec3/resources/definitions/fdmprinter.def.json#L440) is not configured correctly. Please look into your printer's Machine Settings to correctly set the shape of the print head around the nozzle.

LD3D, 2019-08-26:
Never mind you clearly do not comprehend the real issue here at all and 
it is pointless to continue I will just use a different slicer that the 
designers understand issues and will fix them. and not keep blaming the 
user for a defective product. good day to you.