Ultimaker/Cura #6243: [4.2.1] Material retraction changes when creating a new custom FFF

Liger0, 2019-08-23:
**Application version**
4.2.1

**Platform**
Win10 x64

**Printer**
Cr-10S to custom FFF

**Reproduction steps**
In the cr-10S materials, manage your desired material parameter, retraction included

**Actual results**
When creating a new FFF machine, its material are the same except for the retraction lenght, which defaults to 6.5mm for all the materials.

**Expected results**
Keep the retraction lenght used in other machines

**Additional information**
None

Ghostkeeper, 2019-08-23:
I tried your reproduction steps and indeed it reproduces that way.

It seems that the reason for that is that Generic PLA has overrides specifically for Creality CR-10S ([here](https://github.com/Ultimaker/fdm_materials/blob/7fcd73196310b5ce47e72ff05a71d696b2a8a8e1/generic_pla_175.xml.fdm_material#L116)). And by editing the material, you are only editing the override for Creality CR-10S. So if you then switch to Custom FFF Printer, the change is not made for that printer.

However if you make the change again for Custom FFF Printer, then different printers that don't specialise their materials will also have the change. So if you switch to, for instance, Prusa i3, then you'll retain the change because both of those don't have specialised material profiles.

Liger0, 2019-08-23:
I see the retraction amount you linked is 5, that seems different than the 6.5mm I report.

closed 2020-06-22 by nallath