Ultimaker/Cura #6244: [4.2.1] Creality stock profile has a button to force fan speed at 100%

Liger0, 2019-08-23:
**Application version**
4.2.1

**Platform**
Win10 x64

**Printer**
Cr-10S

**Reproduction steps**
Just select a stock profile

**Actual results**
On the fan speed, there is a button to revert to a suggested 100% value instead of the material one

**Expected results**
There should be no button for it.

**Additional information**
That button is very dangerous, because it has no sense as you can force the profile if you want to by yourself, but once clicked that button and saved the profile, the material fan speed is not used anymore and you can't change it back to the material one.

Ghostkeeper, 2019-08-23:
Yeah the fx button reverts it to the most recent formula. This formula for the fan speed sets it to 100% if the fan is enabled or 0% if the fan is disabled. A bit of a weird one since the cool_fan_enabled setting is not a parent setting for the fan speed.

We've been considering to remove the button for all settings and maybe replace it with a button to remove it from the custom profile. That's the use case it was originally intended for anyway.

GregValiant, 2024-11-24:
The post-processor Advanced Cooling Fan Control (in Cura since 5.7.1) gives much finer control of the fan speeds.
Can this Feature Request be closed?

closed 2024-12-09 by github-actions[bot]