Ultimaker/Cura #6245: [master] New crash

smartavionics, 2019-08-23:
**Application version**

master

**Platform**

Linux using QT 5.13

**Printer**

Sorry, can't remember

**Reproduction steps**

Sorry, can't remember, will try to update if it occurs again.

**Actual results**

Cura crashed

**Expected results**

No crash!

**Additional information**

stderr and the log get a large number of the following error reports...

```
2019-08-22 21:56:47,130 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [61]: An uncaught error has occurred!
2019-08-22 21:56:47,133 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]: Traceback (most recent call last):
2019-08-22 21:56:47,134 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]:   File "/opt/cura/lib/python3/dist-packages/cura/CuraApplication.py", line 990, in event
2019-08-22 21:56:47,134 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]:     return super().event(event)
2019-08-22 21:56:47,134 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]:   File "/opt/cura/lib/python3/dist-packages/UM/Qt/QtApplication.py", line 428, in event
2019-08-22 21:56:47,135 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]:     event._function_event.call()
2019-08-22 21:56:47,135 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]:   File "/opt/cura/lib/python3/dist-packages/UM/Event.py", line 189, in call
2019-08-22 21:56:47,135 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]:     self._function(*self._args, **self._kwargs)
2019-08-22 21:56:47,136 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]:   File "/opt/cura/lib/python3/dist-packages/UM/Signal.py", line 315, in __performEmit
2019-08-22 21:56:47,136 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]:     func(dest, *args, **kwargs)
2019-08-22 21:56:47,136 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]:   File "/opt/cura/lib/python3/dist-packages/UM/Backend/Backend.py", line 191, in _onMessageReceived
2019-08-22 21:56:47,136 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]:     self._message_handlers[message.getTypeName()](message)
2019-08-22 21:56:47,137 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]:   File "/opt/cura/lib/cura/plugins/CuraEngineBackend/CuraEngineBackend.py", line 674, in _onGCodeLayerMessage
2019-08-22 21:56:47,137 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]:     self._scene.gcode_dict[self._start_slice_job_build_plate].append(message.data.decode("utf-8", "replace")) #type: ignore #Because we generate this attribute dynamically.
2019-08-22 21:56:47,137 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]: KeyError: 0
2019-08-22 21:56:47,157 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [61]: An uncaught error has occurred!
2019-08-22 21:56:47,157 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]: Traceback (most recent call last):
2019-08-22 21:56:47,158 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]:   File "/opt/cura/lib/python3/dist-packages/cura/CuraApplication.py", line 990, in event
2019-08-22 21:56:47,158 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]:     return super().event(event)
2019-08-22 21:56:47,158 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]:   File "/opt/cura/lib/python3/dist-packages/UM/Qt/QtApplication.py", line 428, in event
2019-08-22 21:56:47,159 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]:     event._function_event.call()
2019-08-22 21:56:47,159 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]:   File "/opt/cura/lib/python3/dist-packages/UM/Event.py", line 189, in call
2019-08-22 21:56:47,159 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]:     self._function(*self._args, **self._kwargs)
2019-08-22 21:56:47,160 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]:   File "/opt/cura/lib/python3/dist-packages/UM/Signal.py", line 315, in __performEmit
2019-08-22 21:56:47,160 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]:     func(dest, *args, **kwargs)
2019-08-22 21:56:47,161 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]:   File "/opt/cura/lib/python3/dist-packages/UM/Backend/Backend.py", line 191, in _onMessageReceived
2019-08-22 21:56:47,161 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]:     self._message_handlers[message.getTypeName()](message)
2019-08-22 21:56:47,161 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]:   File "/opt/cura/lib/cura/plugins/CuraEngineBackend/CuraEngineBackend.py", line 674, in _onGCodeLayerMessage
2019-08-22 21:56:47,162 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]:     self._scene.gcode_dict[self._start_slice_job_build_plate].append(message.data.decode("utf-8", "replace")) #type: ignore #Because we generate this attribute dynamically.
```

smartavionics, 2019-08-23:
Sorry, I should have mentioned that it occurs every now and again. Not all the time.

smartavionics, 2019-08-23:
I also should have mentioned that this has only started occurring in the last couple of days so it's likely that a recent commit is to blame.

Ghostkeeper, 2019-08-23:
I haven't been able to reproduce this so far. I haven't seen it crash in the past couple of days (apart from on my `feature_intent_container_tree` branch that I've been working on).

From the log it looks like the crash happens during the second half of slicing when the engine starts sending layer view data back to the front-end. Maybe that'll help to reproduce it.

smartavionics, 2019-08-23:
My guess from a swift look at the code is that _clearLayerData() is being called while a slice is being done and then when the layer data comes in from the backend there's no longer an entry in gcode_dict.

smartavionics, 2019-08-23:
Possibly found a race. The slice() method contains this sequence...

```
        self._scene.gcode_dict[build_plate_to_be_sliced] = [] #type: ignore #[] indexed by build plate number
        self._slicing = True
        self.slicingStarted.emit()
```

Note that it creates the entry in gcode_dict before it sets _slicing.

if _onStackErrorCheckFinished() gets called between those two statements it can call needsSlicing() and that can call _clearLayerData().

So I will swap those two statements so that _slicing is set before setting the entry in gcode_dict and see if I continue to get the crashes.

smartavionics, 2019-08-23:
I'll submit a PR anyway.

Ghostkeeper, 2019-08-26:
I think it should be really rare if it's a race condition and wouldn't know why it wasn't discovered until now. But I've made a patch that should be thread-safe.

closed 2019-08-26 by Ghostkeeper

smartavionics, 2019-08-26:
OK, I saw your patch. Thanks for that.

I'm guessing that the reason I started seeing the crash recently is because I have been slicing a project that has 4 extruders and so it takes longer for the setting checks to happen?

commit 23f4aa6e (2019-08-26): Fix potential race condition when slice messages arrive after clearing build plate

Fixes #6245.