Ultimaker/Cura #6250: Parametric infill %

kyjak, 2019-08-24:
I would like to be able to set horizontal points in the model and set the infill density.
Between these points, Cura would calculate the infill so that from point A with infill density
of X, would linearly or exponentialy (setting) morph into point B with infill density Y.

Solution is to be able to point a place, can be a dummy model for setting, and specify the
infill %, this done multiple times per model.
Cura then automatically changes the infill geometry so these 2 join without a hard break/wall,
as it is now, rather liearly connect and thicken or thinner the infill with infill lines connected.
I understand, this option would not be available for all infill types (cause of math).

As of now, using multiple models as setting specifiers, one can change the infill and other
parameters in the main model, but with a hard stop/border. Not a fluent transition.

All printers and users affected.

Ghostkeeper, 2019-08-25:
We support something like this, but only for Cross and Cross 3D patterns. Those patterns were actually developed specifically to enable this kind of feature (and it was quite a long research track too).

It's enabled through the setting "Cross Infill Density Image". It allows you to load up an image that will be placed onto the bounding box of your object. Where the image is bright, the infill will be dense. Where the image is dark, the infill will be sparse.

We know that this is not a nice interface to have. It was built that way to first try the functionality before we spend time on making an interface. Now we did think of a more intuitive interface to use, and the way in which we could build that. It would work more like IceSL than what you're proposing. However we haven't had the time yet to implement this.

kyjak, 2019-08-25:
I will try that and report back, thank.you.

kyjak, 2019-08-26:
On the website, there is no help on this setting, and when supplying a BW image path, it got ignored. Tried cross and cross 3d infills. Using 4.2.1, so either it does not work, or something i am doing wrong hmm

Ghostkeeper, 2019-09-02:
Are you sure the path is typed correctly?

It still works for me. This is what happens when I load a cube, Cross infill pattern, 60% infill density, with the Cura logo loaded in:

[image]

kyjak, 2019-09-02:
Yes, path is correct.
Does it need to have a special format or dimensions the image?

Ghostkeeper, 2019-09-05:
It rescales the image to the bounding box of the model, so it should probably be in the same aspect ratio lest you get distortions.

It's using [STB](https://github.com/nothings/stb/blob/master/stb_image.h) to read the images. It supports all of the common raster image formats.