Ultimaker/Cura #6251: Travel move

larsch, 2019-08-24:
**Application version**: 4.2.1

**Printer**
Custom FDM

[travel_moves.3mf.zip](https://github.com/Ultimaker/Cura/files/3537360/travel_moves.3mf.zip)

**Reproduction steps**
Set "Combing" to "Not in skin" or "Inside infill".

**Actual results**
Cura generates travel moves with many jerks, crossing over skin instead of staying inside print when clearly possible.

**Expected results**
Normal travel moves without jerks

**Additional information**
Default slicing where "pointy" travel moves are visible everywhere:

[image]

Example: After printing outer wall, nozzle coasts a bit forward, then jerks outside print, crosses print to other hole in print, then jerks back to next outer wall being printing, even though the inside is fully connected. 

[image]

[image]

smartavionics, 2019-08-24:
Turn off the avoid printed parts when travelling setting.

larsch, 2019-08-24:
Turning off _Avoid Printed Parts_ does prevent the jerk travel moves. OK for this print.

Ghostkeeper, 2019-08-26:
Yeah, looks buggy to me. I can see that it's combing correctly along the first inner wall, but when it needs to separate from that wall it decides to do a `moveOutside` which crosses the wall, and then makes a straight run for the `moveOutside`-location of the destination because it sees that it needs to cross the outer wall anyway (because it's now outside). It shouldn't do that `moveOutside` there.

smartavionics, 2019-08-26:
Yes, it's because it can't comb within the skin. Notice that the spikes don't occur on the layers that don't have skin.

smartavionics, 2019-08-26:
It would be better if it did a non-combed direct travel move with retraction (and, hopefully, z-hop to avoid wrecking the skin).

Liger0, 2019-09-15:
I also have the nozzle going into empty areas going from start point to the ending one on a straight line instead of staying within the model on cura 4.3 beta, no matter if I use non in skin, in infill or all.

smartavionics, 2019-09-15:
Please provide the .3mf file.

Liger0, 2019-09-15:
Literally any model does it.

smartavionics, 2019-09-15:
Unless you provide an example, I cannot help.

Liger0, 2019-09-15:
[Combing.zip](https://github.com/Ultimaker/Cura/files/3614107/Combing.zip)

smartavionics, 2019-09-15:
Thanks for the file. I have sliced it but can't see an obvious problem yet, please say what layer is causing the issue and maybe show an image of the problem. Thanks.

Liger0, 2019-09-15:
I am not very good understanding everything combing does, but I think the travel on the top are not supposed to be there.
[image]

smartavionics, 2019-09-15:
Ah, I see. Thing is, I don't get those travels when I slice with my cura. I think I fixed the bug that causes them a long time ago but the cura developers have not yet incorporated the fix. That was https://github.com/Ultimaker/CuraEngine/pull/973.

[image]

closed 2024-12-07 by github-actions[bot]