Ultimaker/Cura #6252: 4.2.0 Match support extruder to product extruder

frankvdh, 2019-08-24:
**Application Version**
4.2.0
**Platform**
Ubuntu Linux

**Printer**
HE3D sky dual
**Steps to Reproduce**

Print the attached model with two different colours, and support.
[CFFFP_Propeller.zip](https://github.com/Ultimaker/Cura/files/3537519/CFFFP_Propeller.zip)

**Actual Results**
Where the support colour is different from the product, the colour from the support bleeds onto the surface of the product.

**Expected results**
I would like an enhancement giving another option to "Support interface extruder" called "Product extruder". When this is selected, the extruder used to generate the interface layer will vary to be the same as the extruder that will print the corresponding part of the product,

**Additional Information**

Ghostkeeper, 2019-08-25:
One piece of support may support multiple objects though. If you place objects close together the support will merge as per the Support Join Distance. Also when you enable conical support or tree support, one piece of support can print multiple parts. So the support is not particular to any object and can't match the object's extruder.

And I'd argue that this will happen for most multi-colour prints, too. After all, the point of multi-extrusion is that you can have an object with multiple materials in one print, which is only useful if the parts touch. Otherwise you could've printed them separately just as well.

frankvdh, 2019-08-25:
Right, but my proposal relates to the interface layer, not the entire support structure. I don't really care what extruder is used below the interface layer. 

Thinking about how I'd go about this in OpenSCAD, I'd use a Minkowski sum 
I envisage that a shell (including the extruder information) around the product (e.g. Minkowski sum) the thickness of the interface layer would be INTERSECTIONed with the calculated required support to create the interface layer. If that's too computationally expensive, the surfaces of the product could be projected vertically up/down to create the interface layer. Worst case, there may be half an extrusion width of overlap along the edges between the colours, or gaps could be left in the support along these edges.

Here's a photo to show the problem
[image]

Ghostkeeper, 2019-08-26:
> If that's too computationally expensive, the surfaces of the product could be projected vertically up/down to create the interface layer.

This is the implementation we're using. Not just because Minkowski is expensive, but moreso because Minkowski is hard to get working for all types of meshes. There is basically only one proper implementation out there: the one from CGAL, but CGAL is hard to integrate in an existing application like CuraEngine.

However after the projection, we still apply Support Interface Horizontal Expansion. And there may be a different object's interface inside the interface. We need to join these together because otherwise there will be a border between the two.

frankvdh, 2019-08-27:
A border isn't a problem... it's only support, after all. A small gap would be better, and there are already gaps in the support.

Ghostkeeper, 2019-09-02:
The border can be a problem because of printing time and the stability of support if it doesn't adhere well across the border. What's more, it'll be ambiguous where the border lies and that needs to be resolved consistently over every layer even if the layers are processed concurrently on different CPU cores.

It can be made to join only when the different pieces of support have the same extruder, and then only add functionality, in theory. Currently we don't know which piece of support comes from which object (because it's ambiguous). All of the support infill settings are also global at the moment and change depending on the support extruder. We'll have to change the profiles to make support settings per-object.

The ambiguity of where the border lies will look very strange though. Like, which extruder should the base of this support be printed with?

[image]

Ghostkeeper, 2019-09-03:
After discussing this with the team, we've agreed that it's going to be a complex change for relatively little gain, so it's not deserving a high priority. For the time being you could try modelling the support yourself or conceding to use a greater Z distance.

closed 2023-07-13 by github-actions[bot]