Ultimaker/Cura #6253: [4.2.1] Start gcode partially hidden until scroll or window drag

Liger0, 2019-08-24:
**Application version**
4.2.1

**Platform**
Win10 x64

**Printer**
Cr-10S

**Reproduction steps**
Stay in the main cura gui. Open the printers sections, open the cr-10S.

**Actual results**
The start gcode isn't fully displayed until a definite amount of drag or scroll. Only the first line is shown.

**Expected results**
Show the start gcode fully.

**Additional information**
If you just close the printer and reopen it, it will be fixed. But if you close the machine lists, go back to main gui, and reopen machine list, this will trigger again.

https://youtu.be/dzTaUP4WHi8

Ghostkeeper, 2019-08-26:
I'm not able to reproduce this on Ubuntu 19.04. It's displaying fine.

Could be a Qt bug we're running into here. I'm using a more modern version than the Windows build.

Liger0, 2019-09-13:
Okay so on cura 4.3 beta, instead of displaying only the first line in the start gcode field, it doesn't display anything at all.

AbeFM, 2019-09-13:
Try clicking or editing, switching tabs, etc.

I found the text would sometimes appear (and gets put into g-code) even though it doesn't display at first.

Block137, 2019-09-14:
+1
Also had this happen before. No idea how to replicate but it happens quite often (4.1)

Ghostkeeper, 2019-09-17:
So far it only seems to be happening on Windows, but it may also have something to do with the screen DPI.

AbeFM, 2019-09-17:
I generally run a 4k external display, but when I unplug, the native 4k display automatically switches to 2x scaling on the "interface". Not all programs handle this cleanly.

I have no particular reason to suspect this is the issue here, but it's *possible* so I'm mentioning it. I believe I have seen the issue without opening that can of worms.

The other unusual thing I often do is running multiple copies of CURA at once - more than one printer, more than one copy.

closed 2024-12-07 by github-actions[bot]