Ultimaker/Cura #6255: [FR] 3D primitives built-in

Mikerhinos, 2019-08-25:
**Is your feature request related to a problem? Please describe.**

Ok guys let's be honest. We almost all are printing LOTS of calibration tests, and sometimes we only want to test a stupid thing like on a specific corner to check leveling etc...But we don't want to print a model, we just need to check correct height.

**Describe the solution you'd like**

So I thought : why Cura couldn't allow us to add primitives on the bed ? I could make a 0.2mm high cube on top left corner, or a 2cm x 2cm x 2cm cube on center to do a XYZ calibration ? Or even a 300x300x0.2mm cube to test whole bed leveling/heating ?
(It could even become later a calibration tab, with cubes, custom temperature towers, bridge tests etc...).

**Describe alternatives you've considered**

Primitives are...primitive, but creating it the way we want it in Tinkercad or Fusion 360 for example is a bit of a pain because we have to use other tools, export, import, etc...

**Affected users and/or printers**
 
I think that maybe all users would benefit from having 3D primitives built in, and if it can later become a built in benchmarks creator it would be awesome and clearly benefit to all users because each time we buy a new filament brand or type we have to redo a LOT of different tests.
**Additional context**

No additional context, thanks to all the devs who are working on Cura and plugins/mods and letting us using it for free <3

ChrisTerBeke, 2019-08-25:
Seems like a great case for a plugin :)

fieldOfView, 2019-08-26:
I once started this plugin:
https://www.schiedam.nl/a-tot-z/stimuleringsregeling-klimaatadaptatie
As far as I remember, it works with Cura 3.5 and newer, but it is very barebones. If you think it is useful I will consider publishing it on the Marketplace (or including the functionality in the MeshTools plugin).

smartavionics, 2019-08-26:
Hi Aldo, that would be useful. If there was a way to control the resolution of the generated meshes it would be even better.

Mikerhinos, 2019-08-26:
@fieldOfView 
Great, even if it may not very advanced it would be a good start I guess !

Yesterday I watched latest Angus (Makers Muse) video on testing a Silhouette printer, its proprietaty software is based on Cura 2.6, and it has these primitives feature so it should definitely be doable :)

Ghostkeeper, 2019-08-26:
It's definitely possible. In fact, Cura already adds a cube to the scene when you use the Support Blocker tool.

We only have to consider the scope though: Cura is not modelling software. Adding this can open up the scope for "oh but I want a rounded cube" or "why isn't it possible to make a cube with a hole in it" and you'll definitely have people trying to model complete D&D figurines using these primitives and then complaining that the interface is cumbersome to use.

Calibration objects falls slightly on the border of Cura's scope though. It can be considered. But is the interface for loading a calibration object going to be easier than "Click a button on the screen. Select the model you want to load"? Because that's the interface you have right now, except that you need to download the calibration model once first.

Mikerhinos, 2019-08-26:
Didn't thought about these damn users who always want to do what they are not supposed to, lol.
So maybe just cubes with these modifiers :

- Size : X,Y,Z mm
- Position : X,Y mm (not Z as it's not for modeling, it just lies on bed)
- Rotation : degrees around Z (so it will always remain flat on bed)
- Fill : % of X,Y (1% will do a square with line thickness of 1% of X or Y size).

That should permit all the basic leveling tools like a surface covering whole bed, or little squares in angles, a cube in center, towers, lines, etc...
Only the bravest will try to exploit this to model something lol (even if I already imagine some people using 1 layer cubes in multiple projects to change color, and drawing 8-bits style images on bed lol).

diegopradogesto, 2019-08-30:
Hahahaha, a while ago I also started this plugin: https://github.com/diegopradogesto/cura-primitives-extension.

I couldn't find time to continue even if it seems very simple. Maybe soon, at some point I can continue with it. 

For now, using the ThingiBrowser plugin (find it in the Marketplace) could be a good idea. You can search for "calibration" and tons of model will appear that you can just add to the build plate.

Mikerhinos, 2019-08-30:
Yeah but having editable primitives built in would be better IMO.
Like, I changed stock bed of my CR10S Pro to a mirror, that I attached to bed with some 3M adhesive so I can get rid of stock fixations.
Now my print surface is around 310x320mm, and it would be very easy to make custom tests with this kind of tool :)