Ultimaker/Cura #6256: Info needed - Creating new Machine/Variant/Material/Quality

karnalta, 2019-08-25:
Hello,

I am trying since two days to figure out how machine/material/variant/quality are associated but without any luck.

Here is the resources I have created 

[https://wetransfer.com/downloads/ed65ea87b52ef076084f863161f5e43020190825193259/46378c79f56bbe88d76dbe7ce316a70620190825193259/f0862e](url)

It add a new printer under Creality (Ender3 Karnalta) and should have his own global profiles and some specifics profiles for new material/variant association.

I always end up with "Unsupported" as quality profile.. I would really appreciate if someone could tell me what I am doing wrong in my files.

Thank a lot.

nallath, 2019-08-26:
Well, your link doesn't work, so could you update it with one that does have the right files? 

Apart from that, I would recommend installing the godmode plugin (https://github.com/sedwards2009/cura-god-mode-plugin/). It has been a lifesaver for me when debugging profiles.

karnalta, 2019-08-26:
Link is working for me, that's strange.

Here another host : [https://filebin.net/3j7jwaqumd3z5jc8](url)

I will install god mode plugin to give it a try. I will get back here if I find my mistaken by that time.

Ghostkeeper, 2019-08-26:
The original link does work for me except that Github somehow makes it link to "github.com/Ultimaker/Cura/issues/url" which is a broken link. So I have to copy-paste it.

If it says "Unsupported" then there is no quality profile that matches the configuration. Looking quickly at your files (without actually loading them in), there's two things you could try:

* Are you sure you switched to Sunlu ABS rather than Generic PLA? The quality profiles are specialised for that but the preferred material is still Generic PLA.
* Look through the log to see if there are any errors with your files.

I see no obvious mistakes but these things are finicky.

karnalta, 2019-08-26:
Yes I am sure I have switched to my Sunlu material. If it can help, when quality is not supported, the material is highlighted in orange. So I guest Cura want to tell me that this material is not supported for my printer. But there is the definition of my printer in it..

There is no ERROR in the log, basically the log tell me this ;
2019-08-26 11:51:59,681 - DEBUG - [MainThread] cura.Settings.MachineManager._updateQualityWithMaterial [1291]: Updating quality/quality_changes due to material change
2019-08-26 11:51:59,682 - DEBUG - [MainThread] cura.Settings.MachineManager._updateQualityWithMaterial [1298]: Current quality type = [standard]
2019-08-26 11:51:59,684 - INFO - [MainThread] cura.Settings.MachineManager._updateQualityWithMaterial [1307]: No available quality types found, setting all qualities to empty (Not Supported).

Moreover, god mode plugin show me in the extruder stack that my quality profile is loaded but I cannot understand why it's not usable.

karnalta, 2019-08-27:
Ok I have understood where I was messing things up.

If it can help someone else to understand how it work here is what I think.

If the machine definition have specific variant/material quality profiles, it have to have global quality profiles AND variant/material quality profiles. In my tests, if I only let global profiles, I have the "unsupported" problem. If I only let specific variant/material profiles, I still have the "unsupported" problem. I can see profile selection only when both are there.

It took me time to discover this because Cura will keep the name of the global profile and not replace it with the specific profile name. I was using different name to be able to see which profile was selected and it was never changing. But specific settings from the specific profile are in fact applied to the global profile.

So, as in my tests I had just defined a few specifics variant/material qualities among my global profiles, I had mostly unsupported quality except when I was on the right combination of variant/material, where I was seeing my global profile name.

I am not sure I has been really clear but I can't really do better in English :)

Ghostkeeper, 2019-08-29:
Ah, this is expected behaviour then. If the quality profiles are specific to the material and variant then they are also specific to the printer (since the materials and variants are specific to the printer). It will need to have matching quality profiles then.

closed 2019-08-29 by Ghostkeeper