Ultimaker/Cura #6257: Printing PETG no Problem but the result on higher prints is not acceptable

bowie130459, 2019-08-25:
**Application version**

Cura 4.2.1

**Platform**

IMAC 32 GB main Memory 

**Printer**

ENDER 3D Pro
unable to find the file

**Reproduction steps**

[image]
[image]
see picture "Bild Temperaturband 235-195........jpg
Temperature of nozzle is between 235 and 195 and one can see that the fileament is printed to cool. The Temperature from bed (235 degree ) is darke and when it bcomes colder it turns to light green and it breaks.
On the the other photo, after I changed the Temperature via Editor in the gcode file to 220 all over ecpet the bottom it was fine. 

**Actual results**

see above and pictures 

**Expected results**

see bove and pícture 235-220 
 
**Additional information**

In experimentel ther is noc point to mark to keep the Temperature at the value I want to have.

The result is by the moment only reachable while editing the gcode file manually.

Without change CURA isn't usable.

Kind regards
Willi

Ghostkeeper, 2019-08-26:
Do you have a project file that reproduces the issue? I sliced a 10cm high cube with the Creality CR-10 profile and in the g-code it stays at 200 degrees throughout the print.

bowie130459, 2019-08-26:
Attached you will find a Projectfile and the generated GCODE File. M104 changes from botton to top from 235 to 195 Degree as you can see in the GCODE-File.

[CE3_Rdp mitte rechts-2.3mf.zip](https://github.com/Ultimaker/Cura/files/3540897/CE3_Rdp.mitte.rechts-2.3mf.zip)
[CE3_Rdp mitte rechts-2.gcode.zip](https://github.com/Ultimaker/Cura/files/3540898/CE3_Rdp.mitte.rechts-2.gcode.zip)

Kind regards 
WIlli

bowie130459, 2019-08-26:
I closed the issue by wrongly clicking the wrong submit botton

bowie130459, 2019-08-27:
Hi all,

be informed that this issue isn't a bug. A forum entry reminds me to check as if there are preprocesses runing and indeed there was a plugin in place, which I never realised before.
This Plugin, named ChangeszAtZ changes the temperature in a distance in Z as high as the printing was. 
I don't know where and when this plugin was in stalled, but after removing all the scripts no change in temperature took place. So the problem is solved.
But thank you for your assistance.

KInd regards WIlli

closed 2019-08-27 by bowie130459