Ultimaker/Cura #6262: Profile, visibility and material changes after an update

szepeu, 2019-08-26:
**Application version** - 4.0.0 -> 4.2.1

**Platform** - Windows 7 Ultimate 64bit

**Reproduction steps**
 - create a profile on 4.0.0 with some setting over "safe" threshold given by Cura (ie. wipe distance set to 1mm, which for my printer is actually okay)
 - update from 4.0.0 to 4.2.1

**Expected results**
Profiles, materials and visibility settings stay the same after an update

**Actual results**
Some (but not all) parts of the profiles went missing. As an example let's take some settings, which were not set to an unbelievable levels. 25-35mm/s printing speeds, 200-300mm/s travel speeds - some of these values were reset (initial layer print and travel speeds went from 25/200 to 15/80), but outer/inner wall speeds were okay. Wipe distances were reset from 1/0.5mm to 0.2/0.1mm, as well as layer height (0.25->0.15). Infill travel optimization went from true to false. Infill was mostly okay, but gradual infill step was 0.6mm instead of 1mm. Retraction speed and distance is okay, but material temperature went crazy - from 210/60 degree Celcius to 250/75 for generic PLA! Also, all my created materials were deleted. Some of the visibility settings changed, ie. line directions were visible, when i barely use it, thus it was not in the visible settings.
All the profile settings were saved with "Update profile with current settings/overrides" before an update.

**Additional information**
It looks connected to #6125, however not only profile settings changed - materials went totally missing and visibility settings were changed. 
I was worried about installing an update, as in 3.something all profiles were deleted. I installed update to 4.0.0 after a full uninstall from the previous versions to have perfectly clean setup.

Ghostkeeper, 2019-08-27:
Did you ever use Cura 4.1 in between? The first run for Cura 4.2 will update your settings from the latest Cura version that you previously installed. So if you tried 4.1 once, went back to 4.0 and added a bunch of profiles in 4.0, they won't show up in 4.1 and then also not in 4.2. This can be fixed by going to your configuration folder (Help -> Show configuration folder) and deleting the 4.1 and 4.2 folders there. If the 4.2 folder is missing, Cura 4.2 will import again from the latest Cura installation it finds, which is 4.0 then with all your preferences and profiles there.

Otherwise, can we know what printer(s) you were using, and maybe see a log file?

szepeu, 2019-08-27:
Thank You for fast reply. I did not use Cura 4.1 in between, direct update from 4.0.0 to 4.2.1.

I was using my custom 3D printer. I'll upload log file tomorrow when I get to my computer.

szepeu, 2019-08-29:
Here's the log file you requested:
[cura.log](https://github.com/Ultimaker/Cura/files/3554608/cura.log)

Ghostkeeper, 2019-09-02:
Looking at the log, something seems to be going wrong with this Prusa i3 Steel printer. Are you sure that the printer definition is correct and still named the same in the new version? You do seem to have a profile called "Prusa+i3_user.inst.cfg" but it's looking for a profile called "Prusa+i3+Steel_user.inst.cfg". Did the printer's name change?

closed 2020-07-06 by no-response[bot]